INTI-CMNB/KiBot #10: ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:137)

hoijui, 2020-09-09:
Do you have an idea why this might happen?

The project at the relevant commit:
https://github.com/hoijui/for-science-keyboard/commit/bbc5629ba2636f674ded8d29b1d18a02d84b1add

```
Run [ -f *.sch ] && kiplot -d Fabrication -s update_xml,run_drc -i
Using config file: for-science.kibot.yaml
Using SCH file: for-science.sch
WARNING:`update_xml` preflight is not in use, no need to skip (kibot.kibot.kiplot - kiplot.py:191)
- Running the ERC
ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:137)
Traceback (most recent call last):
  File "/usr/bin/eeschema_do", line 574, in <module>
    erc_file = eeschema_run_erc_schematic(output_file_no_ext, eeschema_proc.pid)
  File "/usr/bin/eeschema_do", line 254, in eeschema_run_erc_schematic
    wait_eeschema_start()
  File "/usr/bin/eeschema_do", line 135, in wait_eeschema_start
    wait_eeschema(5)
  File "/usr/bin/eeschema_do", line 104, in wait_eeschema
    return wait_for_window('Main eeschema window', r'Eeschema.*\.sch', time, others=others)
  File "/usr/lib/python3/dist-packages/kicad_auto/ui_automation.py", line 306, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Main eeschema window window
ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:137)
Traceback (most recent call last):
  File "/usr/bin/eeschema_do", line 574, in <module>
    erc_file = eeschema_run_erc_schematic(output_file_no_ext, eeschema_proc.pid)
  File "/usr/bin/eeschema_do", line 254, in eeschema_run_erc_schematic
    wait_eeschema_start()
  File "/usr/bin/eeschema_do", line 135, in wait_eeschema_start
    wait_eeschema(5)
  File "/usr/bin/eeschema_do", line 104, in wait_eeschema
    return wait_for_window('Main eeschema window', r'Eeschema.*\.sch', time, others=others)
  File "/usr/lib/python3/dist-packages/kicad_auto/ui_automation.py", line 306, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Main eeschema window window
ERROR:ERC returned 1 (kibot.kibot.pre_erc - pre_erc.py:45)
##[error]Process completed with exit code 10.
```

hoijui, 2020-09-09:
I managed to reproduce a similar error locally now (the above was on github CI):

```
- 'Print F.Cu+Dwgs.User' (print_front) [pdf_pcb_print]
ERROR:Time-out waiting for pcbnew, giving up (pcbnew_do - pcbnew_do:161)
Traceback (most recent call last):
  File "/usr/bin/pcbnew_do", line 579, in <module>
    print_layers(args.kicad_pcb_file, args.output_dir, args.output_name[0], args.record, args.fill_zones)
  File "/usr/bin/pcbnew_do", line 208, in print_layers
    wait_pcbew_start()
  File "/usr/bin/pcbnew_do", line 159, in wait_pcbew_start
    wait_pcbnew(5)
  File "/usr/bin/pcbnew_do", line 132, in wait_pcbnew
    return wait_for_window('Main pcbnew window', r'Pcbnew', time, others=others)
  File "/usr/lib/python3/dist-packages/kicad_auto/ui_automation.py", line 306, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Main pcbnew window window
```
This error appears when I have the InteractiveHtmlBom plugin installed. The `- Print ...  [pdf_pcb_print]` parts ran fine before installing it. and work fine after removing it again.

set-soft, 2020-09-09:
There are various ways to debug these problems.
The easier is to use `-v` command line option. This will enable verbosity level 1, and pass it to the `kicad-automation-scripts`.
This will save a video containing the KiCad session. 
The best way to watch the video is to decompose it in frames using:

```
ffmpeg -i *.ogv  -vsync 0 out%03d.png
```

In your case looking at the output frames I see:

[image]

If I run:

```
eeschema for-science.sch
```

I get the same error. You must let KiCad import your project to v4 SCH format (you are currently using v2). This will create the nasty `for-science-rescue.lib` and a symbols table `sym-lib-table`. Saving your SCH as KiCad 5.x will solve the problem.

For the BoM you'll need to disable various fields that aren't part of your project. Also note you'll get a warning about `exclude_any`, this option is now part of the filters and variants mechanism. 

I also got a warning about M8 component in PCB but not in the schematic. Nothing really important but you should add it to the SCH.

Here is what I patched to get all the files:

```diff
diff --git a/for-science.kibot.yaml b/for-science.kibot.yaml
index 7b9d8d1..bdbc625 100644
--- a/for-science.kibot.yaml
+++ b/for-science.kibot.yaml
@@ -62,15 +62,15 @@ outputs:
           name: Parte
         - field: Value
           name: Valor
-        - field: manf#
-          name: P/N
-        - field: manf
-          name: Fabricante
-        - field: digikey#
-          name: 'Cod. Digi-Key'
+#        - field: manf#
+#          name: P/N
+#        - field: manf
+#          name: Fabricante
+#        - field: digikey#
+#          name: 'Cod. Digi-Key'
         - Footprint
-        - field: Desc
-          name: Descripción
+#        - field: Desc
+#          name: Descripción
         - 'Quantity Per PCB'
       exclude_any:
         - column: Reference
```

closed 2020-09-09 by set-soft

hoijui, 2020-09-09:
Wow.. you are a machine! :D 
thank you!!

I will try those things. I am not an electrical engineer, and have very little knowledge about electronics and KiCad. I am an Open Software dev, mainly.
Aha... I never opened the SCH file. ;-)
I just chose this simple project, to try adding some meta-data on the silk screen, and to try automatic generation of gerbers, to use it as a sample OSH electronics project for the DIN SPEC 3015.
Let's see if I get there... with this tool and your help, I am able to take a huge leap! :-)

hoijui, 2020-09-09:
Actually... if you could recommend one of your projects maybe, that uses KiBot (and optimally is also used IRL by others), that would probably make for a better example.
Or should I use one of your sample projects you use for KiBot?