INTI-CMNB/KiBot #12: InteractiveHtmlBom pugin causes: "Time-out waiting for pcbnew"

hoijui, 2020-09-09:
Without the plugin, it works fine.

* [with plugin](#with-plugin)
* [without plugin](#without-plugin)

## with plugin

```
$ kibot -s all -v
DEBUG:Python cache dir: /home/hoijui/.local/lib/python3.8/site-packages/kibot/__pycache__ (kibot - __main__.py:223)
DEBUG:Importing from /home/hoijui/.local/lib/python3.8/site-packages/kibot (kibot.kibot.kiplot - kiplot.py:58)
DEBUG:- Importing out_ibom (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_any_layer (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_any_drill (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_pdf (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_gerber (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_excellon (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_ps (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_gerb_drill (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_pcbdraw (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_pdf_pcb_print (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_kibom (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_pdf_sch_print (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_position (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_step (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_svg (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_dxf (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_svg_sch_print (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_hpgl (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_base (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing out_bom (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing pre_base (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing pre_filters (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing pre_ignore_unconnected (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing pre_erc (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing pre_check_zone_fills (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing pre_update_xml (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:- Importing pre_drc (kibot.kibot.kiplot - kiplot.py:62)
DEBUG:Deactivating macros (kibot.kibot.kiplot - kiplot.py:80)
Using config file: for-science.kibot.yaml
DEBUG:Using configuration file: `for-science.kibot.yaml` (kibot - __main__.py:171)
DEBUG:Parsing preflight options: {'run_erc': True, 'run_drc': True, 'check_zone_fills': False, 'ignore_unconnected': False} (kibot.kibot.config_reader - config_reader.py:84)
DEBUG:Parsing preflight run_erc (kibot.kibot.config_reader - config_reader.py:92)
DEBUG:Parsing preflight run_drc (kibot.kibot.config_reader - config_reader.py:92)
DEBUG:Parsing preflight check_zone_fills (kibot.kibot.config_reader - config_reader.py:92)
DEBUG:Parsing preflight ignore_unconnected (kibot.kibot.config_reader - config_reader.py:92)
DEBUG:Parsing global options: {'output': '%f-%i_%r.%x'} (kibot.kibot.config_reader - config_reader.py:100)
DEBUG:Pre-parsing output options for `print_front` (pdf_pcb_print) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `print_bottom` (pdf_pcb_print) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `interactive_bom` (ibom) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `bom_html` (bom) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `bom_csv` (bom) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `gerbers` (gerber) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `excellon_drill` (excellon) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `gerber_drills` (gerb_drill) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `position` (position) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `pcb_top_g` (pcbdraw) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `pcb_bot_g` (pcbdraw) (kibot.kibot.config_reader - config_reader.py:73)
DEBUG:Pre-parsing output options for `step` (step) (kibot.kibot.config_reader - config_reader.py:73)
Using SCH file: for-science.sch
DEBUG:Using schematic: `/home/hoijui/src/for-science-keyboard/for-science.sch` (kibot - __main__.py:148)
DEBUG:Using PCB: `/home/hoijui/src/for-science-keyboard/for-science.kicad_pcb` (kibot - __main__.py:198)
DEBUG:Starting outputs for board /home/hoijui/src/for-science-keyboard/for-science.kicad_pcb (kibot.kibot.kiplot - kiplot.py:216)
DEBUG:Preflight checks (kibot.kibot.kiplot - kiplot.py:172)
DEBUG:Skipping all pre-flight actions (kibot.kibot.kiplot - kiplot.py:176)
DEBUG:Board loaded (kibot.kibot.kiplot - kiplot.py:145)
- 'Print F.Cu+Dwgs.User' (print_front) [pdf_pcb_print]
DEBUG:Output destination: /home/hoijui/src/for-science-keyboard/build (kibot.kibot.kiplot - kiplot.py:202)
DEBUG:Running: ['pcbnew_do', '--version'] (kibot.kibot.kiplot - kiplot.py:89)
DEBUG:PCB title: `for-science` (kibot.kibot.gs - gs.py:129)
[... 171 lines cut ...]
DEBUG:Waiting for b'6293745' window to lose focus... (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:254)
DEBUG:Currently focused id: b'904' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:261)
INFO:Waiting for "Print dialog" ...
DEBUG:Found Print dialog window (1) (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:279)
DEBUG:Window id: b'6292243' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:284)
DEBUG:Waiting for b'6292243' window to get focus... (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:241)
DEBUG:Currently focused id: b'6292243' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:244)
DEBUG:Waiting for b'6292243' window to lose focus... (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:254)
DEBUG:Currently focused id: b'904' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:261)
INFO:Waiting for "Main pcbnew window" ...
DEBUG:Found Main pcbnew window window (1) (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:279)
DEBUG:Window id: b'6291588' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:284)
DEBUG:Waiting for b'6291588' window to get focus... (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:241)
DEBUG:Currently focused id: b'6291588' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:244)
INFO:Exiting pcbnew
DEBUG:Waiting for b'6291588' window to lose focus... (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:254)
DEBUG:Currently focused id: b'6291588' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:261)
DEBUG:Currently focused id: b'904' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:261)
INFO:Retry pcbnew exit
DEBUG:Waiting for b'6291588' window to lose focus... (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:254)
DEBUG:Currently focused id: b'904' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:261)
DEBUG:Terminating the session recorder (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:144)
```

[image]

set-soft, 2020-09-09:
Interesting. I also have IBoM as plug-in.
Are you sure you don't have another KiCad instance running?
Look at the log file:

```
INFO:Waiting for "Main pcbnew window" ...
DEBUG:Found "^Confirmation$" window instead of pcbnew (pcbnew_do - pcbnew_do:146)
```

Here the script informs that `pcbnew` didn't start normally. Instead a dialog with *Confirmation* as title was detected. Is hard to know confirmation for what. Then:

```
INFO:Waiting for "^Confirmation$" ...
DEBUG:Found ^Confirmation$ window (1) (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:279)
DEBUG:Window id: b'6291459' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:284)
DEBUG:Waiting for b'6291459' window to get focus... (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:241)
DEBUG:Currently focused id: b'6291459' (pcbnew_do.kicad_auto.ui_automation - ui_automation.py:244)
INFO:Dismiss pcbnew already running
```

This is just a guess of the script. I can get the *Confirmation* dialog if I already have another instance of `pcbnew` running, with the same PCB opened.
Here the script is just pressing *Enter* to bypass the dialog, in the hope that closing it will allow access to the `pcbnew` GUI.

```
INFO:Waiting for "Main pcbnew window" ...
ERROR:Time-out waiting for pcbnew, giving up (pcbnew_do - pcbnew_do:161)
```

But it doesn't get the window, a dialog still blocking the main Window.

We can do two things to find a solution:

1) You can go to the next debug level: connect to the virtual X server to see what the script is getting.
2) Just try to look for a mechanism to disable the plug-ins, assuming this is something with the plug-in.

If you want to go for option 1:
a) Install x11vnc and a client for it (I use Debian package ssvnc)
b) Get a fresh copy of: https://github.com/INTI-CMNB/kicad-automation-scripts
This is needed because I added a stop point that should help for this problem, the script you have installed will try to close the problematic dialog without stopping.
c) From you project run:
```
__PATH__TO__AUTOMATION__SCRIPTS___/src/pcbnew_do -s -w -r -vv export --output_name for-science-F_Cu+Dwgs_User_.pdf for-science.kicad_pcb build F.Cu Dwgs.User
```
The `-s` option will start the `x11vnc` server and the `-w` option will stop at various points waiting for a key press.
The log messages will inform you that the server was started and will stop waiting for a key. Pressing *Enter* will go to the next wait point.
d) Now run the client, I use:
```
ssvncviewer :0
```
Note that my DISPLAY environment variable is `DISPLAY=":1"` so I use `:0`
Once connected to the `x11vnc`server you'll see the virtual desktop used by KiCad in a window.
With it we should find what the dialog says.

Note that this "display" doesn't have any Window Manager, making interaction a little bit raw. If you want to interact with KiCad (in parallel to the script) you can install `fluxbox` WM and use the `-m` option.

set-soft, 2020-09-10:
Perhaps you can try with `-r` and `-w` using the current git code, the video should include the dialog thanks to the new wait point.

set-soft, 2020-12-09:
I'm closing this issue because I don't have more information and can't reproduce it.
If you have more information please re-open.

closed 2020-12-09 by set-soft