INTI-CMNB/KiBot #14: dnf_filter does not work (or at least not documented) for pick_and_place

justyn, 2020-09-22:
At least for the pick_and_place output (and possibly others) the documentation indicates:

```
dnf_filter: [string|list(string)=''] Name of the filter to mark components as not fitted. A short-cut to use for simple cases where a variant is an overkill.
```

However putting anything in this field leads to:

```
Unknown filter [whatever] used for `dnf_filter`
```

It is not clear how to get the pick_and_place output to ignore DNF components without using variants.

set-soft, 2020-09-22:
Hi @justyn !

You are right about the lack of documentation. Is my fault. I just added a variants example, and not even one filter example.

Right now I'm really busy trying to adapt things so they work for KiCad 6. The current status is that many things are broken in KiCad 5.99, some are being fixed and others don't. So I'm trying to find ways to solve them or the scripts will lose many features when KiCad 6 is released. As an example: I can't get a reliable DRC.

Filters must be defined in a similar way you define a variant. 

The [tests/yaml_samples](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples) directory contains all the regression tests. Many of them test the filters functionality.

- [int_bom_exclude_any.kibot.yaml](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples/int_bom_exclude_any.kibot.yaml): Shows how to use regular expressions to match fields and exclude components. Is the more powerful filter mechanism.
- [int_bom_fil_1.kibot.yaml](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples/int_bom_fil_1.kibot.yaml): Shows various mechanisms. In particular how to change the list of keywords, usually used to match 'DNF', meaning you can exclude components with arbitrary text.
- [int_bom_fil_2.kibot.yaml](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples/int_bom_fil_2.kibot.yaml): Shows how to use KiCad 5 module attributes (from the PCB) to filter SMD, THT and Virtual components. Note KiCad 6 is redefining the attributes.
- [int_bom_include_only.kibot.yaml](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples/int_bom_include_only.kibot.yaml): Shows how to use regular expressions to match only some components, instead of including a few.
- [int_bom_var_t2is_csv.kibot.yaml](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples/int_bom_var_t2is_csv.kibot.yaml): Shows how to use filters and variants simultaneously, not a good idea, but possible.
- [print_pdf_no_inductors_1.kibot.yaml](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples/print_pdf_no_inductors_1.kibot.yaml): Shows how to change the `dnf_filter` for a KiBoM variant.
- [print_pdf_no_inductors_2.kibot.yaml](https://github.com/INTI-CMNB/KiBot/tree/master/tests/yaml_samples/print_pdf_no_inductors_2.kibot.yaml): Shows how to do what `print_pdf_no_inductors_1.kibot.yaml` does but without the need of a variant.

You also have a few built-in filters:

- **_mechanical** is used to exclude:        
  - References that start with # 
  - Virtual components
  - References that match: '^TP[0-9]*' or '^FID'
  - Part names that match: 'regex': 'mount.*hole' or 'solder.*bridge' or 'solder.*jump' or 'test.*point'
  - Footprints that match:  'test.*point' or 'mount.*hole' or 'fiducial'
- **_kibom_dnf_Config** it uses the internal `dnf_list` to exclude components with
  - Value matching any of the keys
  - Any of the keys in the `Config` field (comma or space separated)
- **_kibom_dnc_Config** it uses the internal `dnc_list` to exclude components with
  - Value matching any of the keys
  - Any of the keys in the `Config` field (comma or space separated)

Note that the last two uses a field named `Config`, but you can customise them invoking **_kibom_dnf_FIELD**. This will create an equivalent filter, but using the indicated **FIELD**.

The current list of **DNF** keys is:
- dnf
- dnl
- dnp
- do not fit
- do not place
- do not load
- nofit
- nostuff
- noplace
- noload
- not fitted
- not loaded
- not placed
- no stuff

The current list of **DNC** keys is:
- dnc
- do not change
- no change
- fixed

You can define your own lists as the `int_bom_fil_1.kibot.yaml` shows.

I think you can just use the built-in `_kibom_dnf_Config` filter for simple cases.

Please let me know if you got it working so I can close this issue and move this explanation to the README as a first documentation draft.

justyn, 2020-09-22:
Hi @set-soft 

Thanks very much for your response, and the hard work on this useful tool.

You're right, using `dnf_filter: _kibom_dnf_Config` for pick and place, iBOM etc was exactly what I needed.

The other information on custom filters is also very useful.

I assume there is no way of applying a filter to warnings logged during the outputs process? I see that we can do so for preflight checks, but I am trying to suppress some annoying BOM warnings.

Thanks again.

set-soft, 2020-09-25:
Hi @justyn ! 
Currently the KiBot warnings can't be filtered, I added some code to avoid repetition, but not yet filters.
You can open a separated issue for it so I don't forget about it.

set-soft, 2020-09-28:
Ok, I added the above text and more information about the generic filter options to the README.

Is not a perfect doc, but is enough for a start.

closed 2020-09-28 by set-soft