INTI-CMNB/KiBot #17: pdf_pcb_print lacks output options

stream1972, 2020-10-07:
In current state, pdf_pcb_print is almost useless because it has no options to control output. There are no options to toggle title block, mirror, and scaling, although they exist in KiCad.

It lead to bad situation that there is no good way to create fabrication/documentation outputs:
- Manually in KiCad, I can do almost perfect output -  at fit-to-page scale, excluding title block, mirroring bottom layers. But no support for variants.
- In kibot, variants are handled properly (unfit components are crossed out), but prints itself are at default options, ugly and useless - tiny board inside title block which is impossible to turn off, prints of bottom layers are unreadable because they're not mirrored.

set-soft, 2020-10-07:
Hi @stream1972 !

I use `pdf_pcb_print` output for some real world uses, so I don't agree with the *almost useless* part.
Controlling options from KiCad dialogs is a real nightmare. 
When you open the dialogs which widget is selected isn't always reliable, widgets lack accelerators, their layout changes even for "patch" releases, etc.
Some options can be controlled from the configuration file, if the options you need are controlled there then we can add them.
I'll try to take a look at which options are stored, please try to find it too.

But: from your description it looks like what you want is more compatible with the `pdf` output (or similar).  It has 12 options, including scale, mirror and title-block exclusion (in fact I think the title block can't be included, KiCad limitation in the Python API)

Please let me know which case isn't covered by the `pdf` output to focuse on it.

set-soft, 2020-10-07:
I added support for most options to [KiAuto](https://github.com/INTI-CMNB/KiAuto). 

Lamentably the **mirror** option can't be controlled using the configuration file. In fact this option is lost as soon as you close the dialog. I filled a bug report on KiCad: [5909](https://gitlab.com/kicad/code/kicad/-/issues/5909).

And this dialog is a real mess as I suggested in the previous post. The selected widget depends on things external to KiCad (most probably because KiCad doesn't care about selecting an specific widget), this means that trying to select the **Print mirrored** option will be a problem.

set-soft, 2020-10-14:
I added support for mirror to KiCad 5.99+ 

If you really need to use `pdf_pcb_print` instead of `ps` or `pdf` I'll suggest you to try using https://gitlab.com/ole.tange/tangetools/blob/master/mirrorpdf/mirrorpdf or similar tool.

I'm not including it because it has some heavy dependencies, in particular `pdftk`. Simple solutions, like using `ps-utils` has nasty side effects, like rotating the page.

closed 2020-10-14 by set-soft