INTI-CMNB/KiBot #20: dynamic module does not define module export function

r00tc0d3, 2020-10-19:
Hi,

I did the following on a Mac using:
```
Installation on other targets

Install KiCad 5.x
Install Python 3.5 or newer
Install the Python Yaml module
Run the script src/kibot
```

Resulting in

```
> python --version
Python 3.8.3
```

```
pip show PyYAML
Name: PyYAML
Version: 5.3.1
Summary: YAML parser and emitter for Python
Home-page: https://github.com/yaml/pyyaml
Author: Kirill Simonov
Author-email: xi@resolvent.net
License: MIT
Location: /Users/marcel/.pyenv/versions/3.8.3/lib/python3.8/site-packages
Requires:
Required-by: kibot
```

KiCad
```
Application: KiCad
Version: (5.1.7-0-10_14), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.39.2
Platform: macOS Catalina Version 10.15.7 (Build 19H2), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
    Boost: 1.73.0
    OpenCASCADE Technology: 7.4.0
    Curl: 7.54.0
    Compiler: Clang 10.0.1 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=OFF
    KICAD_USE_OCC=ON
    KICAD_SPICE=ON

```

When running src/kibot I get the following error message.
```
Traceback (most recent call last):
  File "/Users/marcel/KiCad/KiBot/kibot/kiplot.py", line 40, in <module>
    import pcbnew
  File "/Applications/KiCad/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.py", line 15, in <module>
    import _pcbnew
ImportError: dynamic module does not define module export function (PyInit__pcbnew)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./kibot", line 15, in <module>
    from kibot.__main__ import main
  File "/Users/marcel/KiCad/KiBot/kibot/__main__.py", line 71, in <module>
    from .kiplot import (generate_outputs, load_actions, config_output)
  File "/Users/marcel/KiCad/KiBot/kibot/kiplot.py", line 42, in <module>
    log.init(False, False)
TypeError: init() takes 0 positional arguments but 2 were given
```

I can this be resolved? Any ideas?

r00tc0d3, 2020-10-20:
Hi,

I am wondering how this works on Linux (I will try the coming days, when I have de time :-)
KiCad is using python 2.7. On my Mac 2.7.16 to be precise. If I invoke this python version import pcbnew works fine.
When I use (using pyenv) exactly the same python version (2.7.16) on my host import pcbnew gives the above error. However this should also be the case on linux. But in your readme you state to use Python 3.5 or newer. But the _pcbnew.so is "build" for the Kicad python version.
So how does this work on your computer with Python 3.5 or newer together with the 2.7 version of Kicad using the _pcbnew.so binding?

I am not a real python user (mostly because of the version stuff and site packages which don't work and I give up) so I might get some terminology wrong.

set-soft, 2020-12-28:
On Linux KiCad is compiled using Python 3. Using Python 3.x is ok on Linux.

KiCad doesn't have to include Python, the Python interpreter from the OS is used.

The problem is with Windows and Mac. I don't know the reason to keep using an obsolete Python compiler. Python 2.7 was released 10 years ago, and its *end of life* was january 2020.

set-soft, 2020-12-28:
I'll close this issue because I don't see any way to solve the situation.

I can help trying to support Python 2.7, but it won't be easy, the plug-ins mechanism uses too much Python 3 stuff.

closed 2020-12-28 by set-soft