INTI-CMNB/KiBot #21: PCB print leads to `pcbnew_do: error: unrecognized arguments: --scaling --pads`

justyn, 2020-10-22:
I have checked out KiAuto from https://github.com/INTI-CMNB/KiAuto and used the command `alias pcbnew_do=PATH_TO_REPO/src/pcbnew_do` as per instruction (with PATH_TO_REPO replaced).

However when running the latest head of KiBot (67d879e) it prints this error when it gets to pdf printout of the PCB:

```
[pdf_pcb_print]
usage: pcbnew_do [-h] [--record] [--rec_width REC_WIDTH] [--rec_height REC_HEIGHT] [--start_x11vnc] [--use_wm] [--verbose] [--version] [--wait_key] [--wait_start WAIT_START]
                 {export,run_drc} ...
pcbnew_do: error: unrecognized arguments: --scaling --pads
usage: pcbnew_do [-h] [--record] [--rec_width REC_WIDTH] [--rec_height REC_HEIGHT] [--start_x11vnc] [--use_wm] [--verbose] [--version] [--wait_key] [--wait_start WAIT_START]
                 {export,run_drc} ...
pcbnew_do: error: unrecognized arguments: --scaling --pads
ERROR:pcbnew_do returned 2 (kibot.kibot.out_pdf_pcb_print - out_pdf_pcb_print.py:108)
```

I'm not sure what the issue is.

Running `KiAuto/src/pcbnew_do --scaling --pads` just to see what happens, it prints out the same:

```
usage: pcbnew_do [-h] [--record] [--rec_width REC_WIDTH] [--rec_height REC_HEIGHT] [--start_x11vnc] [--use_wm] [--verbose] [--version] [--wait_key]
                 [--wait_start WAIT_START]
                 {export,run_drc} ...
pcbnew_do: error: unrecognized arguments: --scaling --pads
```

set-soft, 2020-10-22:
Hi!
Are you using the `scaling` and `drill_marks` options?
Can you post the .kibot.xml file used?

justyn, 2020-10-22:
Hi @set-soft 

Thanks for looking into this.

I get the same as you when I run `pcbnew_do export --help` and `pcbnew_do --version`.

I don't think I'm using the scaling and drill_marks options, here is the section from my yaml:

```
  - name: 'pdf_pcb_print_top'
    type: 'pdf_pcb_print'
    dir: 'docs'
    options:
      # [string|list(string)=''] Name of the filter to mark components as not fitted.
      # A short-cut to use for simple cases where a variant is an overkill
      dnf_filter: ''
      # [string='%f-%i%v.%x'] filename for the output PDF (%i=layers, %x=pdf). Affected by global options
      #output: '%f-pcb-top.%x'
    layers:
      - layer: F.Cu
      - layer: F.CrtYd
      - layer: Edge.Cuts

  - name: 'pdf_pcb_print_bottom'
    type: 'pdf_pcb_print'
    dir: 'docs'
    options:
      # [string|list(string)=''] Name of the filter to mark components as not fitted.
      # A short-cut to use for simple cases where a variant is an overkill
      dnf_filter: ''
      # [string='%f-%i%v.%x'] filename for the output PDF (%i=layers, %x=pdf). Affected by global options
      # output: '%f-pcb-bottom.%x'
    layers:
      - layer: B.Cu
      - layer: B.CrtYd
      - layer: Edge.Cuts
```

Here is the output using `-vv`:

```
- '' (pdf_pcb_print_top) [pdf_pcb_print]
DEBUG:Output destination: /home/justyn/test/output/built/docs (kibot.kibot.kiplot - kiplot.py:253)
DEBUG:Applying filter `Dummy` to fitted (kibot.kibot.fil_base - fil_base.py:85)
DEBUG:Storing filtered PCB to `/tmp/tmpyjkxyv2c.kicad_pcb` (kibot.kibot.out_pdf_pcb_print - out_pdf_pcb_print.py:68)
DEBUG:Executing: ['pcbnew_do', '-r', '-vv', 'export', '--output_name', '/home/justyn/test/output/built/docs/test-F_Cu+F_CrtYd+Edge_Cuts.pdf', '-f', '--scaling', '1.0', '--pads', '2', '/tmp/tmpyjkxyv2c.kicad_pcb', '/home/justyn/test/output/built/docs', 'F.Cu', 'F.CrtYd', 'Edge.Cuts'] (kibot.kibot.kiplot - kiplot.py:134)
usage: pcbnew_do [-h] [--record] [--rec_width REC_WIDTH] [--rec_height REC_HEIGHT] [--start_x11vnc] [--use_wm] [--verbose] [--version] [--wait_key]
                 [--wait_start WAIT_START]
                 {export,run_drc} ...
pcbnew_do: error: unrecognized arguments: --scaling --pads
DEBUG:Failed with error 2, retrying ... (kibot.kibot.kiplot - kiplot.py:140)
usage: pcbnew_do [-h] [--record] [--rec_width REC_WIDTH] [--rec_height REC_HEIGHT] [--start_x11vnc] [--use_wm] [--verbose] [--version] [--wait_key]
                 [--wait_start WAIT_START]
                 {export,run_drc} ...
pcbnew_do: error: unrecognized arguments: --scaling --pads
ERROR:pcbnew_do returned 2 (kibot.kibot.out_pdf_pcb_print - out_pdf_pcb_print.py:108)
```

Even though `export` is listed after `Executing:` the pcbnew_do error message seems to suggest it is not actually being called with `export`, perhaps?

set-soft, 2020-10-23:
I think I found the problem: the recommendation of using a shell `alias` is ok for running KiAuto manually, but a bad idea for running KiAuto from KiBot.

Do you have another copy of `pcbnew_do` in your path?

What I tried in my system was to rename `/usr/bin/pcbnew_so` to `/usr/bin/pcbnew_so.ok`, defined the alias, tried to generate the output as the debug messages you sent imply (without problems) and then tried KiBot ... it failed to find `pcbnew_do`.

Can you try installing KiAuto? (you can use `pip` or the `setup.py`, you can even use `pip` to install the version from PyPi, is updated, this is because I need a fresh KiAuto for the KiBot regression tests, so KiAuto is realeased very often, I automated the release mechanism, a tag in KiAuto makes a GitHub release, a PyPi release and rebuilds the kicad_auto and kicad_auto_test docker images (both: KiCad 5.1.x and KiCad 5.99) all automatically).

justyn, 2020-10-23:
You're right, installing KiAuto from pip (and then running git checkout of KiBot) did the trick.

Thanks!

closed 2020-10-23 by justyn