INTI-CMNB/KiBot #22: Feature request: ability to set column names in pick and place (position)

justyn, 2020-10-23:
Currently the "position" type for pick and place data fixes the column names as follows:

`Ref,Val,Package,PosX,PosY,Rot,Side`

Some manufacturers do not automatically correctly recognise these headings correctly when importing the data and must be manually changed. For example `PosX` may need to be changed to `Center X`.

If there was an option to allow specifying what each column is called, then it could be adjusted to meet manufacturer requirements.

set-soft, 2020-10-23:
I implemented columns configuration. It passes a very trivial test.

Is a little bit annoying for just renaming a column but allows to remove columns and change the order.

I'm using this as test: https://github.com/INTI-CMNB/KiBot/blob/master/tests/yaml_samples/simple_position_csv_cols.kibot.yaml

closed 2020-10-23 by set-soft

justyn, 2020-10-23:
@set-soft That was fast work, thanks :-)

commit de9628e5 (2020-10-23): Added columns configuration for position files.

You can customize which columns are used, their names and order.
Closes #22