INTI-CMNB/KiBot #23: pdf_pcb_print throws erroneous WARNING:KiCad project file not found

justyn, 2020-10-24:
I've found that pdf_pcb_print is throwing this warning:

```
WARNING:KiCad project file not found
```

This is from [here](https://github.com/INTI-CMNB/KiAuto/blob/721afec7e52cfa037f2741b117dafdbc65e77f41/kiauto/file_util.py#L220)

I had a quick look and it seems that pdf_pcb_print is operating on a temporary file (ie `/tmp/tmp9j_9c3fb.pro`) and so fails the check.

So I think the check needs to be disabled for pdf_pcb_print.

As an extra note, this warning doesn't have a number and so can't be filtered manually either, which might be desirable if you want to run the tool on a kicad_pcb file on its own.

set-soft, 2020-10-24:
As this is a KiAuto error message KiBot knows nothing about it. But we could look for a way to include KiAuto warnings.

I'll try to add the corresponding .pro file to avoid the warning.

set-soft, 2020-10-24:
This patch seems to fix the issue. I added a regression test for it.

closed 2020-10-24 by set-soft

justyn, 2020-10-25:
That fixed it, thanks

commit 68269c10 (2020-10-24): Avoid KiAuto warnings about missing project for pdf_pcb_print

This applies to cases using variants.
Now we copy the current project to the temporal location.
Closes #23