INTI-CMNB/KiBot #24: Incorrect modification of pcb file during (at least) pdf_pcb_print

stream1972, 2020-10-27:
When pdf_pcb_print modifies a PCB to draw crosses for variant, it's deleting some layers from pad layer set.

Sometimes pads have no signal layers at all. Examples of such pads are extra holes for solder paste in QFN packages.

This leads to situation that after modification pad has no layers remaining. An incorrect tag `(layer  )` (without any data) is written to updated PCB file. KiCad will not load such file, it'll generate an error (visible in recorded session):

`footprint U1, pad (unnamed): could not find valid layer for pad`

and either refuse to open the board, either confuse KiAuto strong enough to exit with "Timeout waiting for Print dialog window" message.

To reproduce:
- Start new project;
- Place a component in QFN package on schematic, for example, `AX5043`. 
- Update PCB.
- Assign component to some variant (using `Config` field)
- Create pdf_pcb_print output to print **another** variant (so this component should be crossed out). It'll not work.

Possible solution: if pad layer set becomes empty after modification, don't include this pad record at all.

set-soft, 2020-10-27:
Hi @stream1972 !

Thanks for reporting it!

Looking at the footprint used by AX5043: this is an abnormal situation, a pad that has solder paste, but doesn't have copper, or even worst: has solder mask over it, is a ridiculous pad. Those pads are over another real pad and I can't see their purpose. Seems to be some remainings of the thermal vias.

In any case: this is a real case. Removing the pads to add them again is more complex. So I'm just adding a solder mask aperture.

This seems to solve the problem.

set-soft, 2020-10-27:
The following commit should fix it: b27872d7e8f9670acf3e57af29f73502a7aa1eb2
Re-open it if fails for your case. I added a very simple test using the AX5043 component.

closed 2020-10-27 by set-soft

stream1972, 2020-10-28:
Thanks! It's working now with my big project as well.

I only want to note that such pads are completely valid and widely used. Such QFN packages has a metal bottom which is used  as a big extra pad (usually GND, and sometimes the only GND on the chip). Soldering them is not trivial. A first rule is that solder paste must be **not** placed over whole surface of the pad. Instead, four small drops in the corners must be used. To achieve this. main (copper) pad have no solder paste opening at all, and four additional small pseudo-pads are used to create necessary openings for paste. They have no signal or other layers attached, they only go to gerber files used to create a solder paste stencil.