INTI-CMNB/KiBot #28: Allow for adjustment of rotation in pick and place file

lifeisafractal, 2020-11-29:
Not all fab houses follow the same standards for rotation in the pick and place process. The example that brought this up for me is that KiCad follows a left/upper left Pin 1 convention where as JLCPCB considers 0 degree rotation based on the components are oriented in the tape and reel and these two don't always match. This causes the need for a lot of manual editing to get things right which is time consuming and error prone.

I came across [this blog post](https://dubiouscreations.com/2019/10/21/using-kicad-with-jlcpcb-assembly-service/) which call out a tool (I think written by the author) that will use some filters to automatically adjust the rotation based on pad name. I think this would be a nice feature to add to kibot.

I could see adding a `rotation_fix` entry in the position output. You could create regular expressions to match footprint names and then apply a rotation correction. If just footprint feels a little restrictive, we could add another key called `field` that lets you specifiy the field to filter on. This way you could exclude cretin components from getting modified based on designator or part number.

```yml
- name: 'position_example'
    comment: 'Generates the file with position information for the PCB components, used by the pick and place machine.'
    type: 'position'
    dir: 'Example/position_dir'
    options:
      # [list(dict)] Rotation fixes to apply based on a footprint regular expression match
      # Only the first match is applied.
      rotation_fix:
        - name: Do not correct Q4, this part is special
          field: Reference
          regex: "^Q4$"
          rotation: 0
        - regex: "^SOT-223"
          rotation: 180
        - regex: "^SOT-23"
          rotation: 180
        - regex: "^SOT-353"
          rotation: 180
        - name: All QFNs need to be corrected
          regex: ""^(.*?_|V)?QFN-(16|20|24|28|40)(-|_|$)"
          rotation: 270
...
```

I could take a stab at implementing this, but wanted to check before putting the effort in to see if this has a shoot at getting accepted upstream and to get any feedback on the design.

set-soft, 2020-11-30:
> Not all fab houses follow the same standards for rotation in the pick and place process. The example that brought this up for me is that KiCad follows a left/upper left Pin 1 convention where as [...]

Interesting, didn't know about it.

> This causes the need for a lot of manual editing to get things right which is time consuming and error prone.

Sure.

> I came across [this blog post](https://dubiouscreations.com/2019/10/21/using-kicad-with-jlcpcb-assembly-service/) which call out a tool (I think written by the author) that will use some filters to [...]

Yes!

> I could see adding a `rotation_fix` entry in the position output. You could create regular expressions to match footprint names and then apply a rotation correction. If just footprint feels a little [...]

Ok, I have some questions here:

1) I see JLCKicadTools has a database for the list of parts to fix. How this interacts with the list you mention?
2) How many rotation options are needed? 90, 180, 270?
3) I think this could be implemented using the KiBot filters. So you just say: "Apply XX degrees rotation to the parts passing filter YY". Filters are already implemented and they offer plenty of filtering options. You could also implement a "filter plug-in". I think using the already existing filters is the best way.
4) If this will affect the position, and we are generating position files internally, not using Kicad: I think we could just implement the rotation/position transformation inside KiBot. Avoiding the need for an external tool and extra steps. Do you agree?

> I could take a stab at implementing this, but wanted to check before putting the effort in to see if this has a shoot at getting accepted upstream and to get any feedback on the design.

Ok, I'll wait for answers to the above questions so we can trace a good strategy.

Thanks!

set-soft, 2020-12-28:
Hi @lifeisafractal !

The following commit: 4a3e7fa adds a filter used to achieve rotations.

The examples I used are: [simple_position_rot_1.kibot.yaml](https://github.com/INTI-CMNB/KiBot/blob/master/tests/yaml_samples/simple_position_rot_1.kibot.yaml) and [simple_position_rot_2.kibot.yaml](https://github.com/INTI-CMNB/KiBot/blob/master/tests/yaml_samples/simple_position_rot_2.kibot.yaml)

The only missing part is some documentation.

closed 2020-12-28 by set-soft

daveismith, 2020-12-31:
@set-soft, I'm looking for this exact feature as I implement sphawes/index#43 along with the 3d model support which I saw was just added. Do you have any idea when the next kicad_auto image would be pushed with these updates? It would be fantastic to be able to get this functionality added to the project.

set-soft, 2021-01-04:
Hi @daveismith !  
I released 0.9.0 and the docker images are updating right now.
Please report any glitch.

lifeisafractal, 2021-01-04:
@set-soft This is awesome, that you so much for taking the time to implement this! I'll be able to give the new version a test later this month.

nerdyscout, 2021-01-05:
> Hi @daveismith !
> I released 0.9.0 and the docker images are updating right now.
> Please report any glitch.

I find the readme of kicad_auto irritating, as the docker image is built with all latest packages the tags do match anymore... not sure what's the best way to fix this, maybe just remove it and make a hard link to latest and nightly.

MattCampbellST, 2022-02-01:
@set-soft I just came back to running a new board and used KiBot. The JLCPCB rotation worked great. I just wanted to say thank you for making such a great and useful project, and thanks for getting on KiCad6 support so quickly!

lifeisafractal, 2022-02-01:
@set-soft I just came back to running a board at JLCPCB and wanted to say thanks for making such a great tool for KiCad. It is such a delight to use. Also, thanks for getting on adding KiCad 6 support so quickly!