INTI-CMNB/KiBot #31: Import error when running

skorokithakis, 2020-12-12:
I'm trying to run Kibot on my laptop (after getting it to run with #30 on my desktop), and now I'm getting:

```
21:40:59 $ kibot
ERROR:Unable to import plug-ins: cannot import name 'macros' from 'kibot.macros' (/home/stavros/.local/lib/python3.8/site-packages/kibot/macros.py) (kibot.kibot.kiplot - kiplot.py:64)
```

Unfortunately the import system is so opaque that it takes quite a while to figure out exactly what the problem is, so I can't file a PR at the moment, if anyone knows why this is happening please let me know.

set-soft, 2020-12-12:
How did you install it?
What do you get running `kibot -vv`?
Try removing all `__pycache__` dirs on `/home/stavros/.local/lib/python3.8/site-packages/kibot/`

skorokithakis, 2020-12-12:
Sorry, I found the problem :/ I had forgotten the `--no-compile` option when installing. This error could be improved to print something like "are you sure you installed with `--no-compile`?".

set-soft, 2020-12-16:
Thanks for the idea.
I added a message to remind it.

commit bb3087c9 (2020-12-16): Added a hint for pip installations without using `--no-compile`.

Closes #31
closed 2020-12-16 by set-soft