INTI-CMNB/KiBot #33: A way to pass XML to IBOM automatically

dsa-t, 2020-12-16:
Currently, if I specify `extra_fields`, but not `netlist_file`, the fields do not appear in the ibom output.
The problem is that the path to netlist file is obviously different for different boards, so I cannot use the same config for all.

So it would be nice to have a way to pass the netlist file to IBOM automatically, without modifying the config for each board.

set-soft, 2020-12-16:
This is not a good idea. I don't know the consequences if the XML is out of sync. 
What I can do is to add pattern expansions to it, so you can specify `%F.xml`

set-soft, 2020-12-16:
I implemented the `%F.xml` thing.
Can you test it from the git repo?

closed 2020-12-16 by set-soft

dsa-t, 2020-12-16:
One issue:
if `netlist_file` is not specified or empty, it tries to pass the path to the KiBot working directory and fails with an error.

if `netlist_file: '%F.xml'`, everything works as expected.

commit d201843b (2020-12-16): File name patterns are allowed for the `netlist_file` option.

- Closes #33
- File name patterns: %F is the name of the source file without
  extension, but with the path.