INTI-CMNB/KiBot #34: ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:153

aep, 2020-12-16:
Hi,

i'm trying to automate export of the bom.xml file for further processing.

the file exported by type: bom seems to be something else, so i tried:

preflight:
    update_xml: true

but that just leads to a timeout

set-soft, 2020-12-16:
This could be related to a lot of problems. Generating the XML is something delicated, no simple way, just UI interaction.
Try running `kibot` using `-vv` command line option. 
Post the output here. Also take a look at the generated video.

aep, 2020-12-16:
thanks,

```
INFO:Waiting for "Main eeschema window" ...
DEBUG:xdotool getwindowfocus failed! (eeschema_do.kiauto.ui_automation - ui_automation.py:236)
DEBUG:Time-out waiting for eeschema, will retry (eeschema_do - eeschema_do:121)
ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:153)
DEBUG:Closing pipe with 226112 (eeschema_do.kiauto.ui_automation - ui_automation.py:35)
DEBUG:Terminating 226112 (eeschema_do.kiauto.ui_automation - ui_automation.py:47)
DEBUG:Terminating the session recorder (eeschema_do.kiauto.ui_automation - ui_automation.py:142)
DEBUG:Closing pipe with 226075 (eeschema_do.kiauto.ui_automation - ui_automation.py:35)
DEBUG:Terminating 226075 (eeschema_do.kiauto.ui_automation - ui_automation.py:47)
DEBUG:Restoring old eeschema config (eeschema_do.kiauto.file_util - file_util.py:151)
DEBUG:Restoring old KiCad common config (eeschema_do.kiauto.file_util - file_util.py:151)
DEBUG:Checking if /home/aep/kicad/boards/jlcpcb/jlcpcb.pro was modified (eeschema_do.kiauto.file_util - file_util.py:241)
DEBUG:Failed with error 10, retrying ... (kibot.kibot.kiplot - kiplot.py:140)
DEBUG:Detected KiCad v5.1.8 (5001008) (eeschema_do - misc.py:108)
DEBUG:Eeschema config: /home/aep/.config/kicad/eeschema (eeschema_do.kiauto.file_util - file_util.py:167)
DEBUG:Moving current config to /home/aep/.config/kicad/eeschema.pre_script (eeschema_do.kiauto.file_util - file_util.py:174)
DEBUG:Creating an eeschema config (eeschema_do - eeschema_do:460)
DEBUG:Selecting plot format pdf (4) (eeschema_do - eeschema_do:463)
DEBUG:KiCad common config: /home/aep/.config/kicad/kicad_common (eeschema_do.kiauto.file_util - file_util.py:167)
DEBUG:Moving current config to /home/aep/.config/kicad/kicad_common.pre_script (eeschema_do.kiauto.file_util - file_util.py:174)
DEBUG:Creating a KiCad common config (eeschema_do - eeschema_do:490)
DEBUG:Waiting for virtual X server ... (eeschema_do.kiauto.ui_automation - ui_automation.py:73)
DEBUG:Current DISPLAY is :803527255 (eeschema_do.kiauto.ui_automation - ui_automation.py:74)
DEBUG:Checking using ['setxkbmap', '-query'] (eeschema_do.kiauto.ui_automation - ui_automation.py:84)
DEBUG:Recording session with: ['recordmydesktop', '--overwrite', '--no-sound', '--no-frame', '--on-the-fly-encoding', '-o', '/home/aep/kicad/boards/jlcpcb/bom_xml_eeschema_screencast.ogv'] (eeschema_do.kiauto.ui_automation - ui_automation.py:137)
INFO:Waiting for "Main eeschema window" ...
DEBUG:xdotool getwindowfocus failed! (eeschema_do.kiauto.ui_automation - ui_automation.py:236)
DEBUG:Time-out waiting for eeschema, will retry (eeschema_do - eeschema_do:121)
INFO:Waiting for "Main eeschema window" ...
DEBUG:xdotool getwindowfocus failed! (eeschema_do.kiauto.ui_automation - ui_automation.py:236)
DEBUG:Time-out waiting for eeschema, will retry (eeschema_do - eeschema_do:121)
ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:153)
DEBUG:Closing pipe with 226785 (eeschema_do.kiauto.ui_automation - ui_automation.py:35)
DEBUG:Terminating 226785 (eeschema_do.kiauto.ui_automation - ui_automation.py:47)
DEBUG:Terminating the session recorder (eeschema_do.kiauto.ui_automation - ui_automation.py:142)
DEBUG:Closing pipe with 226784 (eeschema_do.kiauto.ui_automation - ui_automation.py:35)
DEBUG:Terminating 226784 (eeschema_do.kiauto.ui_automation - ui_automation.py:47)
DEBUG:Restoring old eeschema config (eeschema_do.kiauto.file_util - file_util.py:151)
DEBUG:Restoring old KiCad common config (eeschema_do.kiauto.file_util - file_util.py:151)
DEBUG:Checking if /home/aep/kicad/boards/jlcpcb/jlcpcb.pro was modified (eeschema_do.kiauto.file_util - file_util.py:241)
ERROR:Failed to update the BoM, error 10 (kibot.kibot.pre_update_xml - pre_update_xml.py:39)

```

the generated video file is empty (14kb no frames), but i'll try to figure out why

set-soft, 2020-12-16:
Looks like KiCad isn't even starting. Something is failing at low level. 
Make sure the `xdotool` command is installed and working, just try `xdotool getwindowfocus`. The log you sent shows that `xdotool` failed to detect any window in the virtual X server. This is coherent with empty video file.
Also make sure that running `eeschema` from the working directory opens KiCad.

set-soft, 2021-02-09:
Almost 2 months without reply.
I'm closing it, feel free to re-open it if you have more information.

closed 2021-02-09 by set-soft