INTI-CMNB/KiBot #35: KiBoM generation failure

clayton13, 2020-12-23:
Thank you for efforts in aggregating all these tools and making the result incredibility easy to use! 

I am having an issue when using Github Actions with v0.8.1, when I run locally there are no issues and the KiBot completely runs. But on github actions the CSV KiBOM fails while the IBOM and other functionality works. 

From the github action I can extract this:
```
- Updating BoM in XML format
WARNING:(W037) Field Mouser of component R813 (R_US 10k) contains extra spaces: `603-RC0603FR-0710KL ` removing them. (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W037) Field Digikey of component U703 (TLV70233_SOT23-5) contains extra spaces: `296-32415-1-ND ` removing them. (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W041) Missing library `mylib` (kibot.kibot.kicad.v5_sch - log.py:82)
- 'Print schematic (PDF)' (print_sch) [pdf_sch_print]
- 'Interactive Bill of Materials (HTML)' (interactive_bom) [ibom]
- 'Bill of Materials in CSV format' (bom_csv) [kibom]
ERROR:Failed to create BoM, error 1 (kibot.kibot.out_kibom - out_kibom.py:379)
```
https://github.com/INTI-CMNB/KiBot/blob/986f0c715703f6b11c37b2009c94d60c61cac012/kibot/out_kibom.py#L376-L382

I think the issue is pertaining to the missing library. Is the library really required to generate the bom? Can't the cache or rescue libraries be used instead? 

Going forward is there a way to get more verbose information from Github Actions? I see the -v flag but do not know how to pass though the action config or the yaml config.

set-soft, 2020-12-23:
Hi @clayton13 !

> Thank you for efforts in aggregating all these tools and making the result incredibility easy to use!

Thanks!

> I think the issue is pertaining to the missing library. Is the library really required to generate the bom? Can't the cache or rescue libraries be used instead?

I don't know if this is the source of the problem.
You can get more control over the BoM generation using the internal BoM generator (output type `bom`). I suggest using it. It implements all the KiBoM functionality and more.

> Going forward is there a way to get more verbose information from Github Actions? I see the -v flag but do not know how to pass though the action config or the yaml config.

The action was contributed by @nerdyscout and we never finished the integration (my fault).
You are right the `entrypoint.sh` script has a `-v` option, but the `action.yml` doesn't have a mechanism to pass it.
At least for debug I suggest using the mechanism explained in the following example: https://github.com/INTI-CMNB/kicad_ci_test
This is more complex, but you get more control over the parameters.

clayton13, 2020-12-23:
I'll switch my config over to `bom` and report back shortly. 

I have been basing my tests off of @nerdyscout's https://github.com/nerdyscout/ProMicro which does not use the CSV option anywhere. I wonder if they ever had the same issue. 

> At least for debug I suggest using the mechanism explained in the following example: https://github.com/INTI-CMNB/kicad_ci_test

Thanks for pointing out an example. Over the next few days, I'll try to track down the problem using it and provide better logging context to go over.

set-soft, 2021-02-09:
Hi @clayton13 ! Did you manage to solve it?

Bucknalla, 2021-02-21:
Hi @set-soft I am also having this issue trying to run kibot locally, I'm seeing an issue relating to what I think is wxApp and X11 failing to set up correctly (I'm using Ubuntu 20.04)? These are my error logs

```bash
Using SCH file: test.sch
- Updating BoM in XML format
../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
Traceback (most recent call last):
  File "/usr/bin/eeschema_do", line 596, in <module>
    eeschema_bom_xml_commands(cfg)
  File "/usr/bin/eeschema_do", line 419, in eeschema_bom_xml_commands
    wait_for_window('Bill of Material dialog', 'Bill of Material')
  File "/usr/lib/python3/dist-packages/kiauto/ui_automation.py", line 313, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Bill of Material dialog window
../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
Traceback (most recent call last):
  File "/usr/bin/eeschema_do", line 596, in <module>
    eeschema_bom_xml_commands(cfg)
  File "/usr/bin/eeschema_do", line 419, in eeschema_bom_xml_commands
    wait_for_window('Bill of Material dialog', 'Bill of Material')
  File "/usr/lib/python3/dist-packages/kiauto/ui_automation.py", line 313, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Bill of Material dialog window
ERROR:Failed to update the BoM, error 1 (kibot.kibot.pre_update_xml - pre_update_xml.py:39)
```

Any thoughts here would be greatly appreciated!

Thanks :)

set-soft, 2021-02-22:
Hi @Bucknalla !

The message:

```
../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
```

Is a KiCad bug (#6989). KiBot is suppressing it, but the last KiAuto release isn't. The current KiAuto code in git is suppressing the message.

The problem you experiment seems to be related with KiBot failing to open the *Bill of Material* dialog. In order to find what's going on run KiBot with a high verbosity level, use `-vvv`. You should get a lot of information and also a video recording the session. Please post them so we can find the source of the problem.

set-soft, 2021-04-05:
Ok, no extra info and both tools changed.
Closing it, if anybody has more info please reopen it.