INTI-CMNB/KiBot #36: Error while installed deb package on WSL2 / Ubuntu 20.04

NightSkySK, 2021-01-02:
Once I try to install Kibot on Ubuntu 20.04 LTS which is set on WSL2 windows environment I get the following error msg:

```
Setting up kibot (0.8.1-1) ...
Traceback (most recent call last):
  File "/usr/bin/kibot", line 15, in <module>
    from kibot.__main__ import main
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 71, in <module>
    from .kiplot import (generate_outputs, load_actions, config_output)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 47, in <module>
    m = re.search(r'(\d+)\.(\d+)\.(\d+)', pcbnew.GetBuildVersion())
AttributeError: module 'pcbnew' has no attribute 'GetBuildVersion'
dpkg: error processing package kibot (--configure):
 installed kibot package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 kibot
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

set-soft, 2021-01-04:
What KiCad version do you have?
KiCad 5.1.8 has `GetBuildVersion` and I think 5.1.7 too. Not sure which version introduced it.

NightSkySK, 2021-01-04:
@set-soft you're right I used not updated repo and had installed Kicad 5.1.5, after upgrade to 5.1.9 seems to be working.
The issue can be closed. Thank you for support.

closed 2021-01-04 by NightSkySK