INTI-CMNB/KiBot #37: kicad-git-filters.py: No such file or directory

TheSlowGrowth, 2021-01-02:
v0.8.1 runs into this error when executing the Docker image:

`/entrypoint.sh: line 144: /usr/bin/kicad-git-filters.py: No such file or directory`

TheSlowGrowth, 2021-01-02:
I was able to resolve the missing kicad-git-filters.py, see here: https://github.com/TheSlowGrowth/KiBot/blob/feature/install-kicad-git-filters/Dockerfile.

I'm still getting the `TypeError: can only concatenate str (not "NoneType") to str` error.

TheSlowGrowth, 2021-01-02:
The error was ill-formatted KiBot config file (type key is given, but value is empty):

```
[...]
outputs:
  - name: 'Generate schematic PDF'
    comment: "Generate schematic PDF file"
    type: 
    dir: pdf
```

set-soft, 2021-01-04:
Thanks! The action still experimental.

closed 2021-01-04 by set-soft

set-soft, 2021-01-04:
I also added a catch for the empty type so you get a more descriptive message