INTI-CMNB/KiBot #39: pcbnew import error

eeintech, 2021-01-05:
I'm so used to using virtual environment that I created one to install kibot and got this error:
``` bash
$ kibot
ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to PYTHONPATH? (kibot.kibot.kiplot - kiplot.py:43)
ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to PYTHONPATH? (kibot.kibot.kiplot - kiplot.py:43)
```

After some research and tries, it seems like isolating it from the system Python is what causes `pcbnew` to fail the import.
Maybe it would be good to mention this in the README? :smile:

set-soft, 2021-01-05:
Thanks! 
I never use virtualenvs.

closed 2021-01-05 by set-soft

eeintech, 2021-01-05:
That's fine, at least it is mentioned somewhere for anyone running into the same issue 😃

commit 9ce0f891 (2021-01-05): Added a note about virtualenv to the docs.

Closes #39