INTI-CMNB/KiBot #40: Corrupted example.kibot.yaml?

eeintech, 2021-01-05:
I have KiBot up and running on Ubuntu 18.04, here is what I get after generating the example KiBot config:
``` bash
$ kibot -x
Creating /<my-path>/./example.kibot.yaml example configuration

$ kibot --list
Using config file: example.kibot.yaml
ERROR:Error loading YAML while parsing a block mapping
  in "example.kibot.yaml", line 6, column 3
expected <block end>, but found '<scalar>'
  in "example.kibot.yaml", line 19, column 9 (kibot.kibot.error - error.py:40)
```

The issue is with those lines in the `example.kibot.yaml` config files:
``` yaml

preflight:
  ...
  # [boolean=false] Update the XML version of the BoM (Bill of Materials).
        To ensure our generated BoM is up to date.
        Note that this isn't needed when using the internal BoM generator (`bom`)
  update_xml: true
```

After commenting them:
``` yaml
preflight:
  ...
  # [boolean=false] Update the XML version of the BoM (Bill of Materials).
  # To ensure our generated BoM is up to date.
  # Note that this isn't needed when using the internal BoM generator (`bom`)
  update_xml: true
```

It works fine!

eeintech, 2021-01-05:
And btw the new error means this section:
``` yaml
outputs:
    ...
    options:
      # [list(dict)|list(string)] List of columns to display.
      # Can be just the name of the field
      columns:
        # [string=''] Name of the field to use for this column
        field: ''
        # [list(string)|string=''] List of fields to join to this column
        join: ''
        # [string=''] Name to display in the header. The field is used when empty
        name: ''
```

needs to change to:
``` yaml 
outputs:
    ...
    options:
      # [list(dict)|list(string)] List of columns to display.
      # Can be just the name of the field
      columns:
        # [string=''] Name of the field to use for this column
        - field: ''
        # [list(string)|string=''] List of fields to join to this column
        - join: ''
        # [string=''] Name to display in the header. The field is used when empty
        - name: ''
```

I'm still getting some errors...

``` bash
$ kibot --list
Using config file: example.kibot.yaml
Available actions:

Pre-flight:
- check_zone_fills: True
- filters: True
- ignore_unconnected: False
- run_drc: True
- run_erc: True
- update_xml: True
Outputs:
ERROR:In section 'bom_example' (bom): Missing or empty `field` in columns list ({'field': ''}) (kibot.kibot.error - error.py:40)
```

Would actually be nice if the `example.kibot.yaml` file was ready to use :smiley:

set-soft, 2021-01-05:
Well, in fact isn't ready for use ;-) 
Is just an autogenerated template.
But I agree we must fix as much errors as possible

eeintech, 2021-01-05:
Ha right, that makes more sense :+1:

set-soft, 2021-01-05:
Now the file passes the `--list` test, but now I put it clear that this is just a template

closed 2021-01-05 by set-soft

eeintech, 2021-01-05:
@set-soft Cool thanks!

commit a6c4bd3b (2021-01-05): Made clear that generic_plot.kibot.yaml is just a template.

Closes #40