INTI-CMNB/KiBot #43: Is there a way to point to a KiBOM config file?

eeintech, 2021-01-05:
I'm currently trying to generate BOMs using KiBOM and I see a `config.kibom.ini` file being generated.
The problem is KiBot never reads this file, it just overwrites it everytime it runs.
By any chance, is there a way to tell KiBot to use a pre-defined KiBOM config file?

Something in those lines could work:
``` bash
$ kibot -s all -d Generated -g kibom_config=bom.ini kibom_example
```

set-soft, 2021-01-06:
You have to set the `conf` option to the name of the file, and avoid using all the options you see in the template example.

Here is what the doc says:
```
[string|dict] BoM configuration file, relative to PCB.
                You can also define the configuration here, will be stored in `config.kibom.ini`
```

closed 2021-01-06 by set-soft

eeintech, 2021-01-06:
Thanks, it works. It was definitely not obvious based on the doc.

Also I've noticed when defining the variant in the command line option, the variant name is **not** appended to the resulting KiBOM file name...

So `$ kibot -s all -d Generated -g variant=PROD kibom` results in `test_project-bom.csv` instead of `test_project-bom.csv_PROD.pdf`.