INTI-CMNB/KiBot #44: Support for list type in variant option?

eeintech, 2021-01-05:
I'm testing with two variants: `DEV` and `PROD`.

Following the [variant documentation](https://inti-cmnb.github.io/kibot_variants_arduprog/), I've setup my `example.kibot.yaml` as follow:
``` yaml
variants:
  - name: 'default'
    comment: 'Default Variant'
    type: kibom
  - name: 'DEV'
    comment: 'Development Variant'
    type: kibom
    file_id: _DEV
    variant: DEV
  - name: 'PROD'
    comment: 'Production Variant'
    type: kibom
    file_id: _PROD
    variant: PROD
```

and the `output` to:
``` yaml
outputs:
  - name: 'pdf_sch_print_example'
    comment: 'Exports the PCB to the most common exhange format. Suitable for printing.'
    type: 'pdf_sch_print'
    dir: 'Example/pdf_sch_print_dir'
    options:
      # [string|list(string)=''] Name of the filter to mark components as not fitted.
      # A short-cut to use for simple cases where a variant is an overkill
      dnf_filter: ''
      # [string='%f-%i%v.%x'] filename for the output PDF (%i=schematic %x=pdf). Affected by global options
      output: '%f-%i%v.%x'
      # [string=''] Board variant to apply.
      # Not fitted components are crossed
      variant: 'DEV'
  - name: 'pdf_sch_print_example'
    comment: 'Exports the PCB to the most common exhange format. Suitable for printing.'
    type: 'pdf_sch_print'
    dir: 'Example/pdf_sch_print_dir'
    options:
      # [string|list(string)=''] Name of the filter to mark components as not fitted.
      # A short-cut to use for simple cases where a variant is an overkill
      dnf_filter: ''
      # [string='%f-%i%v.%x'] filename for the output PDF (%i=schematic %x=pdf). Affected by global options
      output: '%f-%i%v.%x'
      # [string=''] Board variant to apply.
      # Not fitted components are crossed
      variant: 'PROD'
```

Running `kibot -s all -d Generated pdf_sch_print_example` works, it creates PDF files for both `DEV` and `PROD` variants.

Now, let's say we moved to production and want only the 'PROD` variant, I've tried to run KiBot with the `-g` argument:
``` bash
$ kibot -s all -d Generated -g variant=PROD pdf_sch_print_example
```

Unfortunately, it still creates two PDF files instead of just the one for the `PROD` variant...
I don't know a better way than just removing the output for `DEV` from the `example.kibot.yaml` file, is there any other?

Could the better way be to support a list in the `variant` option of each output?
Something in those lines (similar to KiBOM but in YAML format):
``` yaml
outputs:
  - name: 'pdf_sch_print_example'
      ...
      variant: 
        - 'DEV'
        - 'PROD'
```

Then it just a matter of updating this variant list to `- 'PROD'` only.

There is a chance I'm not using this right from the start, let me know!

set-soft, 2021-01-06:
Using:

`$ kibot -s all -d Generated -g variant=PROD pdf_sch_print_example`

Should work if you define only one output without any specific variant.

You are naming two outputs with the same name, this is a very bad idea, you should name them with different names. Then you can choose which one you want from the command line.

closed 2021-01-06 by set-soft

eeintech, 2021-01-06:
@set-soft I understand your point but why having to add more text to your configuration file when just a new line for each variant is needed? Do you see my point?

You could also generate multiple outputs using the command line, similarly to the config file:
``` bash
$ kibot -s all -d Generated -g variant=DEV,PROD pdf_sch_print_example
```

Also, why did you close this issue? We can still have a conversation based on my initial thoughts, does not feel very professional...