INTI-CMNB/KiBot #45: Linux Mint install failure

ukoda, 2021-01-26:
I have tried to install the kibot_0.9.0-1_all.deb on a couple of Linux Mint 20.1 (Ulyssa) workstations.  One was a freshly set up system with minimal addtional packages installed.  KiCAD 5.1.5 was installed using the package manager.

The error I am seeing is:

david@xps:~$ kibot 
Traceback (most recent call last):
  File "/usr/bin/kibot", line 15, in <module>
    from kibot.__main__ import main
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 71, in <module>
    from .kiplot import (generate_outputs, load_actions, config_output)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 47, in <module>
    m = re.search(r'(\d+)\.(\d+)\.(\d+)', pcbnew.GetBuildVersion())
AttributeError: module 'pcbnew' has no attribute 'GetBuildVersion'
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 153, in apport_excepthook
    with os.fdopen(os.open(pr_filename,
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_usr_bin_kibot.1000.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/kibot", line 15, in <module>
    from kibot.__main__ import main
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 71, in <module>
    from .kiplot import (generate_outputs, load_actions, config_output)
  File "/usr/lib/python3/dist-packages/kibot/kiplot.py", line 47, in <module>
    m = re.search(r'(\d+)\.(\d+)\.(\d+)', pcbnew.GetBuildVersion())
AttributeError: module 'pcbnew' has no attribute 'GetBuildVersion'

I have tried searching for existing solutions with out luck.  The above error is seen when I try to run kibot with no arguments but also appeared when I installed git using apt.

I apologise if this the wrong place to request help but could not see anywhere else that seemed appropriate.

paulsykes, 2021-01-26:
Same problem here. My KiCad Version information is below.
Version: 5.1.5-52549c5-84-ubuntu18.04.1, release build

NightSkySK, 2021-01-26:
This is the same issue as described in issue #36 
You need update your kicad installation up to 5.1.7 or later
[KiCad download page](https://kicad.org/download/linux-mint/)
```
sudo add-apt-repository --yes ppa:js-reynaud/kicad-5
sudo apt update
sudo apt install kicad --install-recommends kicad
# If you want demo projects
sudo apt install kicad-demos
```

paulsykes, 2021-01-26:
I added the following repo from the official kicad install documentation https://kicad.org/download/ubuntu/
```
sudo add-apt-repository --yes ppa:kicad/kicad-5.1-release
```
And then updated by installed packages
```
sudo apt update
sudo apt-get upgrade
```

NightSkySK, 2021-01-26:
@paulsykes Does your Kicad was upgraded to 5.1.9? on this way

set-soft, 2021-01-26:
Thanks for reporting @ukoda and @paulsykes !
And thanks @NightSkySK for pointing in the right direction!
I updated the Debian control file to indicate KiCad 5.1.6 as minimum version.
I also added code to catch the error and generate a warning asking for a newer version.

closed 2021-01-26 by set-soft

ukoda, 2021-01-27:
Thanks for the prompt response.  Just confirming that switching to the kicad repo to get a more recent version has resolved the issues I was seeing on a couple systems that were having issues.

commit 409df8cc (2021-01-26): Fixed minimum KiCad version required is 5.1.6.

- KiCad 5.1.5 lacks GetBuildVersion()
- Fixes #45