INTI-CMNB/KiBot #48: timeout in DRC on fairly complicated board

TomKeddie, 2021-02-06:
Hi, Thanks for an awesome tool.

I'm seeing a timeout running DRC on a 4 layer board (https://github.com/TomKeddie/prj-artix7-pcie/tree/master/hardware/v2).
```
  File "/usr/local/lib/python3.8/dist-packages/kiauto/ui_automation.py", line 313, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Report completed dialog window
```
I couldn't see an option to extend this time, if I hack the script as below it passes (with expected warnings).  If we don't already have one, perhaps we need an option to extend the timeout for more complicated boards (and/or slower PCs).
```
def wait_for_window(name, window_regex, timeout=20, focus=True, skip_id=0, others=None):
```

```
tom@z400:~/git/TomKeddie/prj-artix7-pcie/hardware/v2$ kibot --version
KiBot 0.9.0 - Copyright 2018-2020, Salvador E. Tropea/INTI/John Beard - License: GPL v3+```

set-soft, 2021-02-06:
You are right, the KiAuto time-out can't be set from KiBot.

`pcbnew_do` accepts a `--wait_start` command line option. Can you test if this option is enough? From what you say we need another switch to control the regular time-out, not just the one we wait until KiCad starts (which is worst for each new release).

I'll add such a parameter, but I need more details.

TomKeddie, 2021-02-06:
Hi @set-soft 

> `pcbnew_do` accepts a `--wait_start` command line option.

This doesn't help, it seems it's the regular time-out. 
```
tom@z400:~/git/TomKeddie/prj-artix7-pcie/hardware/v2$ pcbnew_do --wait_start 999 run_drc artix7-pcie-v2.kicad_pcb tmp/
../src/common/stdpbase.cpp(62): assert "traits" failed in Get(): create wxApp before calling this
{'current_version_release': '0.0.4.6'}
Traceback (most recent call last):
  File "/usr/local/bin/pcbnew_do", line 702, in <module>
    run_drc(cfg)
  File "/usr/local/bin/pcbnew_do", line 374, in run_drc
    run_drc_5_1(cfg)
  File "/usr/local/bin/pcbnew_do", line 306, in run_drc_5_1
    wait_for_window('Report completed dialog', 'Disk File Report Completed')
  File "/usr/local/lib/python3.8/dist-packages/kiauto/ui_automation.py", line 313, in wait_for_window
    raise RuntimeError('Timed out waiting for %s window' % name)
RuntimeError: Timed out waiting for Report completed dialog window
```
Given changing the default in wait_for_window() fixes the problem I guess that's where the current default comes from? Perhaps it can be supplied just for the DRC case but I don't know the call hierachy.

Thanks for your help.

set-soft, 2021-02-06:
I added a command line option to `pcbnew_do` and `eeschema_do`: `--time_out_scale`
Using `2.0` should do what you want.

From KiBot you can add:
```yaml
global:
  kiauto_time_out_scale: 2.0
```
Let me know if you can test it.

TomKeddie, 2021-02-07:
> Let me know if you can test it.

@set-soft thanks for being super responsive. I'm hitting the limits of my python knowledge in running this from the repo. I'll keep working on it.  Please know I appreciate you helping so quickly and I'll get back to you as soon as I can.

TomKeddie, 2021-02-07:
@set-soft thank you for the enhancement, this works very well for me.  (I had to install kicad-automation-scripts manually so I could run kibot from source).

set-soft, 2021-02-08:
Thanks @TomKeddie for testing it.
I added a warning suggesting to use the new options when a time-out is detected.

commit ce0f22ed (2021-02-08): Adjusted some details about the time-out control

- Warning when detected pointing to use the global options
- Now pcbnmew_do/eeschema_do output is captured
- Warning when using floating value for kiauto_wait_start (no fail)
- A simple test for time-out
Fixes #48
closed 2021-02-08 by set-soft