INTI-CMNB/KiBot #49: Interactive HTML Bom plug in not found

ukoda, 2021-02-07:
I have the Interactive HTML Bom plug in installed and it working correctly from inside the KiCAD application.  However KiBot throws this error:

ERROR:No `generate_interactive_bom.py` command found.
Please install it, visit: https://github.com/INTI-CMNB/InteractiveHtmlBom (kibot.kibot.kiplot - kiplot.py:126)

The install instructions suggest running:
import pcbnew
print(pcbnew.PLUGIN_DIRECTORIES_SEARCH)
And inside KiCAD this shows the path where I have installed the plugin, which is ~/.kicad_plugins.

I tried modifying kiplot.py with extra output:

def check_script(cmd, url, version=None):
    import pcbnew
    print('Plugins directories = [' + pcbnew.PLUGIN_DIRECTORIES_SEARCH + ']')
    if which(cmd) is None:

The relevant output before the error is thrown is:
Plugins directories = []

Please let me know if you need any further information.

Thanks
  David

set-soft, 2021-02-07:
You have to install iBoM not just as a plug-in but as a tool.

If your system accepts Debian packages just install the package from: https://github.com/INTI-CMNB/InteractiveHtmlBom

ukoda, 2021-02-07:
I followed the wiki install instructions to the releases page and saw three assets there which is where i got the plugin zip file.  I didn't see any reference to a package file in those instructions.  For a Debian package I was expecting a deb file but I don't see with assets.  Am I looking in the wrong place?

set-soft, 2021-02-07:
Hi @ukoda !

You are most probably looking in the original project, not my fork. Here is the releases: https://github.com/INTI-CMNB/InteractiveHtmlBom/releases

ukoda, 2021-02-07:
Yes, that looks more helpful.  It looks like a link needs updated in the documentation.  The chain I followed was:

1. KiBot error message with link https://github.com/INTI-CMNB/InteractiveHtmlBom
2. Installation and Usage section pointing to wiki with link https://github.com/openscopeproject/InteractiveHtmlBom/wiki
3. How to Install section pointing to releases with link https://github.com/openscopeproject/InteractiveHtmlBom/releases
I guess it is that last link that needs updated.

I moved the files from the original link out of the plugin folder then installed the Debian package.  This results in KiCAD no longer offering the Interactive BOM generation so I guess I really need to install the plug in from the old link and the package to get this feature in both KiCAD and KiBot.

Running KiBot again it is still showing that directory list as empty but throws a different error:

Plugins directories = []
ERROR:In section 'kibom' (kibom): Option `columns` must be any of ['list(dict)', 'list(string)'] not `dict` (kibot.kibot.error - error.py:40)

Looking at the output folders it has generated the interactive bom file so the link you have provided has resolved this issue and the error I am now seeing is a new issue that may be my kibot.yaml file still needing further attention.

Restoring the folder from the old link to the plug got KiCAD offering the ability to generate an interactive bom from KiCAD.  There may be some merit in create an equivalent zip file in the assets from the new link but I'm guess the documentation would need updated to reflect the different roles the two different assets play as a newbie like myself is going to assume they both do the same thing different ways.

In summary this issue is fixed for me but but the documentation probably needs updated so others don't make the same mistakes as me.

set-soft, 2021-02-08:
Hi @ukoda !

You are right, the documentation is from the original project and needs update.

About: 
> I moved the files from the original link out of the plugin folder then installed the Debian package. This results in KiCAD no longer offering the Interactive BOM generation so I guess I really need to install the plug in from the old link and the package to get this feature in both KiCAD and KiBot.

How did you install KiCad? In my Debian system, using Debian or Ubuntu packages, KiCad supports loading plugins from: `/usr/share/kicad/scripting/plugins/`
The Debian package installs `InteractiveHtmlBom` in `/usr/share/kicad/scripting/plugins/InteractiveHtmlBom/`. So you should get the `generate_interactive_bom.py` **and** the plug-in from the Debian package.

> In summary this issue is fixed for me but but the documentation probably needs updated so others don't make the same mistakes as me.

I'm adding a note to the docs.
But I'm also adding a mechanism to run the `generate_interactive_bom.py` from the plugin.

commit f698e446 (2021-02-08): Now we can load InteractiveHtmlBom even when installed only as plugin

Fixes #49
closed 2021-02-08 by set-soft