INTI-CMNB/KiBot #50: pcbDraw error :  Unable to determine pcbdraw version.

JulienDevillers, 2021-02-08:
Hello,

with ubuntu 20.04

I installed kibot and pcbdraw like this:
```
python -m pip install kibot
python -m pip install pcbdraw
```

Now, launching kibot with pcbdraw output leads to this error:

```
ERROR:Unable to determine pcbdraw version:
 (kibot.kibot.kiplot - kiplot.py:110)

```

What should I do?

Thanks
Julien

NightSkySK, 2021-02-08:
Please take a look at issue #45 https://github.com/INTI-CMNB/KiBot/issues/45. You've probably has older version of KiCad

JulienDevillers, 2021-02-08:
thanks for your answer but unfortunately, I'm runing 5.1.9
Since your message, I did remove kicad via apt and reinstall via apt ; the result is the same.

```
Application: KiCad
Version: 5.1.9-73d0e3b20d~88~ubuntu20.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.68.0 OpenSSL/1.1.1f zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3
Platform: Linux 5.4.0-65-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.68.0
    Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

```

set-soft, 2021-02-08:
Hi @JulienDevillers !
The problem is that you installed the regular pcbdraw. Try installing the patched version: https://github.com/INTI-CMNB/PcbDraw

Also: if you can install the Debian package do it. Pip can't solve dependencies outside the pip repo. 
As the README explains you'll need 

```
Also install the convert (from imagemagick) and rsvg-convert (from librsvg2-bin) tools.
```

Installing the Debian package will pull these dependencies.

closed 2021-02-08 by set-soft