INTI-CMNB/KiBot #52: Install error 0.10.0-2 AttributeError: 'StreamHandler' object has no attribute 'setStream'

paulsykes, 2021-02-16:
Version 0.9.0-3 working great, tried to update to 0.10.0-2, fails during install on Ubuntu 18.04

```
$ sudo apt install ./kibot_0.10.0-2_all.deb 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'kibot' instead of './kibot_0.10.0-2_all.deb'
The following packages were automatically installed and are no longer required:
  docutils-common python-babel-localedata python3-alabaster python3-babel python3-docutils python3-imagesize python3-jinja2 python3-markupsafe python3-roman sphinx-common
Use 'sudo apt autoremove' to remove them.
Recommended packages:
  kibom.inti-cmnb pcbdraw
The following NEW packages will be installed:
  kibot
0 upgraded, 1 newly installed, 0 to remove and 62 not upgraded.
Need to get 0 B/249 kB of archives.
After this operation, 1,071 kB of additional disk space will be used.
Get:1 /home/paul/Downloads/kibot_0.10.0-2_all.deb kibot all 0.10.0-2 [249 kB]
Selecting previously unselected package kibot.
(Reading database ... 330020 files and directories currently installed.)
Preparing to unpack .../kibot_0.10.0-2_all.deb ...
Unpacking kibot (0.10.0-2) ...
Setting up kibot (0.10.0-2) ...
Traceback (most recent call last):
  File "/usr/bin/kibot", line 15, in <module>
    from kibot.__main__ import main
  File "/usr/lib/python3/dist-packages/kibot/__main__.py", line 72, in <module>
    logger = log.init()
  File "/usr/lib/python3/dist-packages/kibot/log.py", line 126, in init
    ch.setStream(sys.stdout)
AttributeError: 'StreamHandler' object has no attribute 'setStream'
dpkg: error processing package kibot (--configure):
 installed kibot package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 kibot
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

set-soft, 2021-02-16:
Hi @paulsykes !

Thanks for reporting. My fault, I didn't notice that StreamHandler.setStream was introduced in Python 3.7 and the minimal dependency we ask is Python 3.6.

The v0.10.0-3 should fix it. BTW: I think Ubuntu 18.04 has both Python 3.6 and 3.7. The docker images I'm using has 3.7 (Debian buster) and 3.9 (Debian bullseye).

closed 2021-02-16 by set-soft

paulsykes, 2021-02-16:
@set-soft many thanks. Just tested the 0.10.0-4 package. Everything working great on Ubuntu 18.04 now. Have a good evening.

commit 95b48ab0 (2021-02-16): Fixed problem using Python 3.6

- Fixes #52
- StreamHandler.setStream introduced in 3.7