INTI-CMNB/KiBot #53: Unexplained Time Out

joelbrunetti, 2021-02-21:
I am having problems running KiBot using Gitlab CI.

I know my [kibot.yaml](https://gitlab.com/joelbrunetti/heated-socks-battery-hardware/-/blob/3-add-ci-checks-generation/kicad_ci.kibot.yaml) is valid because I can pull the setsoft/kicad_auto:latest Docker image and run all of the kibot commands from by gitlab-ci.yml file to generate my output files.

I know my [gitlab-ci.yml](https://gitlab.com/joelbrunetti/heated-socks-battery-hardware/-/blob/3-add-ci-checks-generation/.gitlab-ci.yml) is valid because the script do execute but fail.

An example of the failure when the script runs on the Gitlab CI Runner:

> Running with gitlab-runner 13.9.0-rc2 (69c049fd)
>   on docker-auto-scale 72989761
>   feature flags: FF_GITLAB_REGISTRY_HELPER_IMAGE:true
> Preparing the "docker+machine" executor 00:31
> Using Docker executor with image setsoft/kicad_auto:latest ...
> Pulling docker image setsoft/kicad_auto:latest ...
> Using docker image sha256:ac7637e4ecae41781c2e99b022c458231f30de074c3bb6fa6a7715de321384d4 for setsoft/kicad_auto:latest with digest setsoft/kicad_auto@sha256:5f73825ba33e1df618e766a86004e008bf19e4377618a9f5f131e6c50403419f ...
> Preparing environment 00:03
> Running on runner-72989761-project-24167928-concurrent-0 via runner-72989761-srm-1613881361-c3130243...
> Getting source from Git repository 00:02
> $ eval "$CI_PRE_CLONE_SCRIPT"
> Fetching changes with git depth set to 50...
> Initialized empty Git repository in /builds/joelbrunetti/heated-socks-battery-hardware/.git/
> Created fresh repository.
> Checking out cda63971 as 3-add-ci-checks-generation...
> Skipping Git submodules setup
> Executing "step_script" stage of the job script 02:47
> Using docker image sha256:ac7637e4ecae41781c2e99b022c458231f30de074c3bb6fa6a7715de321384d4 for setsoft/kicad_auto:latest with digest setsoft/kicad_auto@sha256:5f73825ba33e1df618e766a86004e008bf19e4377618a9f5f131e6c50403419f ...
> $ [ -f *.sch ] && kibot -d report/ -s run_drc -i
> Using config file: kicad_ci.kibot.yaml
> Using SCH file: HeatedSocksBattery.sch
> WARNING:(W008) Unable to find KiCad configuration file (/root/.config/kicad/kicad_common) (kibot.kibot.kicad.config - log.py:82)
> WARNING:(W011) Missing default symbol library table (kibot.kibot.kicad.config - log.py:82)
> - Running the ERC
> ERROR:Output from command:
> > ERROR:Time-out waiting for eeschema, giving up (eeschema_do - eeschema_do:156)
> >  (kibot.kibot.kiplot - kiplot.py:146)
> ERROR:ERC returned 10 (kibot.kibot.pre_erc - pre_erc.py:58)
> Cleaning up file based variables 00:01
> ERROR: Job failed: exit code 1

The duration for this to execute is 3 minutes and 27 seconds.

So I know:
- My kibot.yaml file is correct and does what I want.
- My gitlab-ci.yml file is correct and does what I want.
- The kicad_auto:latest (as of a couple days ago) functions without errors.

I am left suspecting the Gitlab CI servers. The duration seems very high and I suspect KiBot is taking longer to execute on the Gitlab CI servers than on my test machine and thus timing out.

My questions to you are:
- How do I verify my suspicion?
- If I'm correct, how do I change the timeout to be more forgiving of the shared CI runner?
- If I'm wrong, any ideas?

I'm relatively new to KiCad but your project is great. Thank you for the lovely automation tools.

set-soft, 2021-02-22:
Hi @joelbrunetti !

You are right the GitLab CI servers has issues running heavy tasks like KiCad. They worked quite well, but some months ago started to fail. I thought it was a temporal problem (too loaded) and moved the examples to GitHub. 

I'm investigating the problems. I got some advances on this. But the problems are a little bit confusing. I'm working on this repo: https://gitlab.com/set-soft/kicad-ci-test

Take a look at it. As I'm still working you'll need to check which pipelines succeed. To test the last advances I'm doing a `git clone` of KiAuto and KiBot

```yaml
  script:
    - "free -h"
    - "mkdir -p Fabrication"
    - "cat /proc/cpuinfo"
    - "git clone https://github.com/INTI-CMNB/KiAuto.git"
    - "git clone https://github.com/INTI-CMNB/KiBot.git"
    - "export PATH=`pwd`/KiAuto/src:`pwd`/KiBot/src:$PATH"
#    - "KiAuto/src/eeschema_do -r -vvv run_erc -o Fabrication/kicad_ci_test-erc.txt kicad_ci_test.sch Fabrication"
    - "[ -f *.sch ] && kibot -vvv -d Fabrication -s update_xml,run_drc -i"
    - "sync"
    - "ls -la Fabrication"
    - "vmstat -s"
    - "uptime"
```

The last KiAuto release sends the stderr/stdout of its child processes to `/dev/null` to avoid confusing messages. The current git version stores them in separated log files when you ask for verbose levels bigger than 2 (3 in this test). Looking at the logs I discovered that the most common problem is KiCad (recordmydesktop) failing to connect to Xvfb (the virtual X server). This is really confusing because KiAuto checks `xvfb` is usable using `setxkbmap -query`. If this call fails it waits some time and retries. On GitLab CI servers I'm getting success from the `setxkbmap -query` (meaning the X server is up and running) but then `recordmydesktop` says it can't connect to the X server. The display reported by `recordmydesktop` is correct. KiCad reports it couldn't initialize GTK+. So the current code checks if KiCad prematurely finished, in this case waits some time and retries. Three retries are done. This is starting to work, now I must try removing the debug options to see if it continue working.

I suggest you trying to use a similar `gitlab-ci.yml` file and check is it works for you. Note that I'm currently forcing the artifacts collection even on fail, so I can analize all the generated logs. Also note that I'm logging information about the CPU, memory and system load. I first suspected it was heavy load and/or low memory, but the logs shows plenty of free memory and normal loads.

set-soft, 2021-02-22:
Ok, I finished my workaround for GitLab CI issues. The new releases pass my two tests:
- https://gitlab.com/set-soft/kicad-ci-test-spora
- https://gitlab.com/set-soft/kicad-ci-test
- 
Don't know what's going on, it looks like *black magic*, but if you:
1. Wait half a second after detecting Xvfb is fully running.
2. Record the session with `recordmydesktop`

KiCad can consistently connect to the Xvfb server. This is ridiculous because once we can communicate to Xvfb I don't see why we need a delay before `recordmydesktop` can connect, and what is even worst: why do we need to start `recordmydesktop` in order to make KiCad able to connect to the X server.

I  never observed this problem using my machine (Debian buster and bullseye, running normally and using docker images) nor on GitHub CI. And this problem was introduced by some change in the GitLab CI servers between april 25 and august 25 last year.

Anyways, the tests are working now. If you find problems re-open the issue and we'll analyze the problem.

closed 2021-02-22 by set-soft

joelbrunetti, 2021-02-24:
That was incredibly quick. Thank you so much for your help and for this excellent project.

I'll keep an eye to see if Gitlab breaks things again. It is quite unusual.