INTI-CMNB/KiBot #56: Windows support

Freshdouble, 2021-03-05:
Is there currently a support for windows for kibot?

I can install kibot using pip but it can't import pcbnew.

set-soft, 2021-03-12:
Hi @Freshdouble !

Lamentably we don't have Win32 support.
The main problems are:
- KiCad uses Python 2.7 on Windows. This version ended its life cycle in january 2020, after almost 10 years. Adding support for Python 2.7 is hard. In particular because we use some Python features that mess with low level details.
- For various things (ERC, DRC, printing the schematic and PCB (not plotting)) we use KiCad itself. So we emulate the user interaction. Porting this to Windows means somebody must find tools to replace the ones we use and debug it for Windows.

KiCad 6 is switching to Python 3.x and the DRC can be done from Python. I hope the ERC will be also available. I think that once KiCad 6 is released, becomes stable and KiBot supports it, we will be able to have support for Win32 with most of the features working (not the printing stuff, only plot).

So currently you have two options:
1. Use GitHub/GitLab CI/CD services to run KiBot on their servers
2. Use GNU/Linux

closed 2021-03-12 by set-soft

Cushychicken, 2022-04-29:
Sort of a fundamental question, but: does KiBot run natively on Linux machines?

I experience the same issue as the original author and run Windows. 

I'm assuming that if you run the following from a terminal on a Debian/Ubuntu machine with Kicad installed:

```python
$ python -c "import pcbnew"
```

...that the command will "just work", and the package is imported.

I suppose what I'm asking is: does KiCAD use the native/local Python build on Linux machines, instead of having a bundled Python2.7 like on Windows?

set-soft, 2022-04-30:
Hi @Cushychicken: On Debian/Ubuntu (and installing the .deb) KiCad uses the Python installed at system level.

KiCad 5 and 6 for Linux needs Python 3.x at system level.

On Debian/Ubuntu the Python 3.x interpreter is named `python3`, and `python` is the Python 2.7 interpreter.

So KiBot (and KiCad) uses `python3`. Try:

```shell
$ python3 -c "import pcbnew"
```

Cushychicken, 2022-05-01:
That's really convenient - I may have to consider spinning up a Linux image for that. 

In the meantime thanks for answering my question.

Cushychicken, 2022-10-11:
Appreciate the response - I think I understand what's different about this
instead of Windows.

dzid26, 2023-07-07:
Right click Start -> Windows Terminal Elevated:
```
cd "C:\Program Files\KiCad\7.0\bin\Scripts"
.\Scripts\pip install --no-compile kibot
```
Run from anywhere:
```
& "C:\Program Files\KiCad\7.0\bin\Scripts\kibot.exe" --help
```