INTI-CMNB/KiBot #59: Change 3D model with filter or variant

Freshdouble, 2021-03-20:
Is it possible to change the 3D Model of some components based on a filter which can be applied to a variant?

Lets say we have a board with different connectors, which share the same footprint but have different 3D models. Is it possible to change the displayed 3D models for this connectors or should there be two footprints on the board which are overlapping and we filter out all footprints that don't have the right 3D model?

set-soft, 2021-03-31:
Hi @Freshdouble !
Currently we don't have a mechanism to change the 3D model.
I'm committing a very experimental patch to allow it.
This patch works looking for a *magic* schematic field named `_3D_model`. If a component contains this field the 3D model is replaced using the content of the field.
Note that both models must have the same position, scale and rotation. We only change the filename. The StepUp freecad plug-in can be used to force it.
Also note that when using more than one 3D model the `_3D_model` field must contain the same number of filenames, using the `,` as separator.
Now the variants integration: you must add to your variant a `pre_transform` filter of type `var_rename`. This filter must change fields named `VARIANT:_3D_model` to `_3D_model` for the variant that needs to change the 3D model.
I created a very silly test case in the `tests/yaml_samples/step_variant_1.kibot.yaml` test.
The case is really silly because is using `IBoM` variants and a negative match (blacklist), so I had to force a variant name called `default`. Using KiBoM variants should be simpler.
Let me know if you can make it work and if it solves your problem.

Freshdouble, 2021-04-02:
Thank you for your answer and assistance. I will try it the next week or so.

set-soft, 2021-05-04:
Hi @Freshdouble !

Did you tested it?

set-soft, 2021-10-04:
I'm assuming this is working, feel free to open an issue if it has some problem.

closed 2021-10-04 by set-soft