INTI-CMNB/KiBot #60: Support option to invert `rot_footprint` result for bottom mounted components

frankleonrose, 2021-03-22:
I'm generating files for PCB:NG and for bottom mounted parts, the rotation correction (apparently) needs to be `180 - comp. footprint_rot`. Unless I'm mistaken, there's no way to express that in the `rot_footprint` filter.

https://github.com/INTI-CMNB/KiBot/blob/master/kibot/fil_rot_footprint.py#L94-L97
```
                if self.negative_bottom and comp.bottom:
                    comp.footprint_rot -= angle
                else:
                    comp.footprint_rot += angle
```

A new option (`invert_bottom`?) would allow it. `negative_bottom` allows subtraction, `rot - angle`, then `invert_bottom` would make that `angle - rot`.
```
                if self.negative_bottom and comp.bottom:
                    comp.footprint_rot -= angle
                else:
                    comp.footprint_rot += angle
                if self.invert_bottom and comp.bottom:
                    comp.footprint_rot = -comp.footprint_rot
```

Is there a better way to do this? Would you accept a PR?

set-soft, 2021-03-31:
Hi @frankleonrose !
Yes, create a PR.
Thanks

set-soft, 2021-04-08:
Hi @frankleonrose !
Thanks for the PR, I reviewed it, take a look at the needed changes.

closed 2021-04-25 by set-soft

set-soft, 2021-04-25:
Hi @frankleonrose !

I merged your PR with some changes. In particular the `skip_top` and `skip_bottom` options, instead of changing the names of the footprints.

As you are working with `PCB:NG`: It would be very interesting if you can comment on which settings are needed for this manufacturer. Ideally you could create a sample template like this: `doc/samples/JLCPCB.kibot.yaml`

These files help to get fabrications files ready for known manufacturers.