INTI-CMNB/KiBot #62: (Feature request) Integrate "-e" CLI flag into yaml file

sethkaz, 2021-03-24:
So, the -e flag is used to specify the schematic file to pick when the KiCAD project has more than 1 schematic sheet.

Would it be possible to include this information in the yaml file instead?  

I'm envisioning something under "kibot:" or "preflight:" that would be "schematic_root: top_sheet.sch", so we can just type in kibot into the command line to run it.

closed 2021-04-01 by set-soft

set-soft, 2021-04-01:
I think we don't really need it.
I made the schematic detection smarter.
Try now, you can give kibot the following clues:

1. Name the configuration file using the same base name as your top schematic.
2. Make sure you have a project file with the same base name.
3. Create a PCB file with the same base name. Note that KiCad 5.1.9 creates it as soon as you create a project.

I think the cases where kibot will fail to pick the right schematic are really strange, and forcing to use `-e` for these cases isn't asking too much.
With the last changes the only case I think that can confuse kibot is: 2 or more projects in the same directory and the configuration file isn't named similar to the corresponding project. And in this case you can rename the configuration file to make things more clear. 
Note that in the case the config is shared by more than one project found in the same directory using something inside it won't solve the problem.
Another case: the config file is outside the project directory. In this case you can use a symlink to clarify.
Let my know if you find a case that can't be solved with the current mechanism.

sethkaz, 2021-04-01:
I guess what I'm more asking for is a way to suppress the warning.  

It is good to be **informed** of which schematic file the software is grabbing, but in the way I'm using it, clue #2 will always be true.  As you pointed out, KiCAD creates the schematic with the project, and names them the same.  I don't change that because it really seems unnecessary.  

The smarter schematic detection has prevented errors or mishaps.  kibot has been getting the correct top-level schematic since version 0.7.0.

I assumed that the best or easiest way would to suppress that warning be to include a flag within the yaml file.  

Alternatively, maybe change that to "INFO:....." so it doesn't look like something is wrong?

set-soft, 2021-04-22:
Hi @sethkaz !

Sorry for the big delay, the e-mail I used for github notifications was failing.

> I guess what I'm more asking for is a way to suppress the warning.
...
> I assumed that the best or easiest way would to suppress that warning be to include a flag within the yaml file.

You can supress any warning from the config file: https://github.com/INTI-CMNB/KiBot#filtering-kibot-warnings

commit dc7ef486 (2021-04-01): Made the schematic name detection smarter.

Should Fix #62