INTI-CMNB/KiBot #64: Does this work with KiCAD 5.99?

G-Pereira, 2021-04-02:


set-soft, 2021-04-02:
Hi @G-Pereira !   
Only a few outputs currently support 5.99.
I advanced on many fronts some months ago by I then stopped because:

1. KiCad 5.99 changes too often, generating all kind of incompatibilities. Keeping the pace is too much work.
2. The schematic format radically changed. The roadmap says KiCad 6 will implement Python bindings for eeschema. So I have two possible roads: a) wait until python bindings are available or b) implement the new format. I chose the first option in order to save effort. After months without advances in the python bindings I'm not sure it was the best. But in this way I hope to avoid "reinventing the wheel".

The KiCad team is changing a lot of things in v6, and they don't provide any kind of compatibility. So KiCad 6 will break most of the plug-ins. KiBot relies on various tools that will need adaptation.

If you want to experiment you can:

1. Use a KiCad nightly package. These packages installs the Python bindings in `/usr/lib/kicad-nightly/lib/python3/dist-packages`
2. Define the `KIAUS_USE_NIGHTLY` environment variable to be `5.99`

But you'll need to mess with the code. I'm not testing any of the migrated features regularly and, as I mentioned before, things changes all the time.

closed 2021-04-02 by set-soft

G-Pereira, 2021-04-03:
Thank you for the datailed response! That makes complete sense, it is indeed constantly changing.