INTI-CMNB/KiBot #65: Run "Annotate Schematic"

thejpster, 2021-04-06:
I have a KiCad project that uses some generic sheets I pull in from another repository as a git submodule. Unfortunately, before I can work on my project in a fresh checkout, I have to run "Annotate Schematic" to get all the page numbers and symbol references sorted out. I can't commit these changes to the other repository containing the generic sheets, as they are specific to my project.

Is it possible to get KiBot to run the "Annotate Schematic" function before doing the ERC or Plot functions?

set-soft, 2021-04-08:
Hi @thejpster !
I can add something to KiAuto, but anything involving dialogs is problematic.
And I see a potential problem here: How do you plan to move to a PCB?

thejpster, 2021-04-08:
I just open the PCB file and say import netlist from schematic, and it's fine. All my parts appear,  whether they are in my 'local' schematic or one of the linked 'generic' subsheets.

So ... I guess I'll also need an option to update the netlist automatically.

Or I guess I can just copy/paste the subsheets into my project repo and commit them with their new project-specific page numbers.

The repo is GitHub.com/Neotron-compute/Neotron-Pico. Look in the feature branch as that's where I'm working.

set-soft, 2021-04-08:
> I just open the PCB file and say import netlist from schematic, and it's fine. All my parts appear, whether they are in my 'local' schematic or one of the linked 'generic' subsheets.
> 
> So ... I guess I'll also need an option to update the netlist automatically.

You can update the netlist using KiAuto.

> Or I guess I can just copy/paste the subsheets into my project repo and commit them with their new project-specific page numbers.
> 
> The repo is GitHub.com/Neotron-compute/Neotron-Pico. Look in the feature branch as that's where I'm working.

Ok, but the main point here is: hardware is much more rigid than software. You can use a git repo, but with some strong limitations.

Once you create the PCB, changing the schematic in the subproject won't automagically change it. You'll need a lot of human work. So I don't see the point in skipping a step that needs human intervention.

I took a look at your project and the problem is the fact that you are reusing `dac4.sch`. One possible solution is to avoid the reuse. This will solve the problem. And I think this is the best way to solve it. In this way you'll get a consistent annotation between both projects.

Another option is: 
1. Create a fork of `Neotron-Common-Hardware`
2. Use this fork in the `Neotron-Pico` project
3. Now you can annotate the project from `Neotron-Pico` and commit the changes to `Neotron-Common-Hardware`'s fork.
Changes to `Neotron-Common-Hardware` will need to be incorporated to the fork when you think this is needed. And you'll need to be very careful with the annotation. Every change in the `Neotron-Common-Hardware` will need a careful work.

The main point is: even git submodules are kind of static, you need to manually ask for refresh. When using hardware this is even worst. Once you create the PCB changes in the schematic means a lot of work, you can't just update the submodule. So I don't see why do you want to get an ERC check on something that really needs human review.

BTW your problem is related to the way KiCad solves the schematics reuse. What KiCad needs implies modifications to `dac4.sch`. Making them temporal and automatic will most probably mean the annotated components will have pseudo-aleatory references. Something that you don't really want.

To make things worst: KiCad 5 handles sheet numbers in a very poor way. The use you are doing will generate sheet number inconsistencies. It means that things like asking for annotation starting with the sheet number multiplied by 100 won't always work  as expected.

closed 2021-04-08 by set-soft

thejpster, 2021-04-13:
I appreciate you taking them time to check out my project in detail. I'm clearly expecting KiCad to be much more like my software dev tools than it actually is, and I will take on-board all your comments. Thank you!