INTI-CMNB/KiBot #69: [FEATURE REQUEST] Generate ".brd" file

joaoantoniocardoso, 2021-04-19:
Hi!

Would it be possible to integrate one more tool into this toolbox? I think exporting ".brd" files would be very interesting. Quoting directly from [this tool](https://github.com/whitequark/kicad-boardview) which I am currently using:

> The KiCAD to Boardview exporter reads KiCAD PCB layout files and writes Boardview files. Boardview is a set of file formats used in the industry for communicating important data about PCB in post-production stage, such as component, pin, and test point placement.
> 
> The Boardview file formats do not typically have a specification or even a name. The KiCAD to Boardview exporter writes ".brd" files with ASCII Boardview data. These files can be opened with MIT-licensed OpenBoardView.

Sincerely I don't really know how many people would be benefited from it and my final intent would be to use it through [this other tool](https://github.com/nerdyscout/kicad-exports) that would enable us to automatically generate the board view right from the Github Actions. I actually post [this issue there](https://github.com/nerdyscout/kicad-exports/issues/24) but it makes more sense to integrate it here, not directly there.

Let me know what you think about it,
Thanks!

closed 2021-04-20 by set-soft

set-soft, 2021-04-20:
Hi @joaoantoniocardoso !
I implemented a `boardview` output. Is 90% copy & paste from the tool you mention. 
Can you test it?
BTW: Do you know who is @whitequark ? I would like to give correct attribution for this portion of code, not just a nick name.

whitequark, 2021-04-20:
I am me. "whitequark" is my name by which people know me in my private and professional life.

joaoantoniocardoso, 2021-04-20:
Hi @set-soft, nice work, for sure I can help testing it, not exactly now but at least tomorrow I give you a return.

joaoantoniocardoso, 2021-04-21:
Hi @set-soft, to test I just ran:

```
kibot -c KiBot/tests/yaml_samples/boardview.kibot.yaml -b my_pcb.kicad_pcb -d out
```
with some of my PCBs and everything seems to be fine! 
Thanks! :100:

commit cfe61c9b (2021-04-20): Added board view export.

Closes #69