INTI-CMNB/KiBot #70: Malformed wire ERROR

CristobalFerrer, 2021-04-20:
Hi, 

While using Kibot to generate my project output files I ran into this error:

`ERROR:At line 304 of `/home/cferrer/Desktop/PCBs/ANYmal_Hardware/anymal_030301_cx_power_cewitch/safety_switch.sch`: Malformed wire (kibot.kibot.kiplot - kiplot.py:202)
ERROR:Line content: `Wire Notes Line rgb(255, 0, 0)` (kibot.kibot.kiplot - kiplot.py:203)`    

It seems Kiplot doesn't support colored Wire Notes Line/Graphic Polyline.
I traced back the error to line 1174 in the v5_sch.py file, https://github.com/INTI-CMNB/KiBot/blob/1d44a6a5514d4f35d87e5d9b9814bafd3e821786/kibot/kicad/v5_sch.py#L1174-L1176

Because "Wire Notes Line rgb(255, 0, 0)" have an extra argument specifying the line color, the length doesn't match 3. 
I fixed the issue by adding an additional condition:
`        if len(res) != 3:
            if len(res) == 6:  
                if "rgb" in res[3]:  
                    pass
                else:
                    raise SchFileError('Malformed wire', line, f)
            else:
                raise SchFileError('Malformed wire', line, f)
`

It would be nice if colored Wire Notes Line would be supported.

set-soft, 2021-04-22:
Hi @CristobalFerrer !

Thanks for reporting. Didn't know it was posible to change color, width and aspect of "wire notes line".
I added support for it.
Can you test it?

closed 2021-04-22 by set-soft

CristobalFerrer, 2021-04-22:
@set-soft will do, thanks a lot!
And thank you also for the amazing job done with the development of Kibot!

commit e22e1db4 (2021-04-22): Added support for width, style and color in "wire notes"

Fixes #70