INTI-CMNB/KiBot #71: directory of kibot.errors.filter

nerdyscout, 2021-04-25:
Hi @set-soft 

according to the readme the kibot_errors.filters should be created in the output dir.
> Important note: this will create a file named kibot_errors.filter in the output directory.

but, from what I could see, kibot is creating this file in the kicad project dir. As those could be different directories I am currently running into issues.

set-soft, 2021-04-25:
Hi @nerdyscout !
The problem you mention in edit 2 should be fixed in 111b86a3751b4e6466c68fdcc60a48bb427e9f77
Are you using git code?

nerdyscout, 2021-04-25:
> The problem you mention in edit 2 should be fixed in [111b86a](https://github.com/INTI-CMNB/KiBot/commit/111b86a3751b4e6466c68fdcc60a48bb427e9f77)
thank you for your quick support.

> Are you using git code?
not sure if I got your question right... I am currently depending on your release packages, but would prefer to run on master. But so far I have not been able to figure out how to implement this in my current setup.

set-soft, 2021-04-25:
The fix isn't released yet.
Using the git code is a simple solution.
I'm working on a new release, including experimental KiCost support.

set-soft, 2021-04-25:
Hi @nerdyscout !

I released 0.11.0, and the docker images are updated.
This should fix your problem.

closed 2021-04-25 by set-soft