INTI-CMNB/KiBot #72: GitHub Action does not appear to be published to GitHub Marketplace

jameswilddev, 2021-05-02:
The documentation indicates that this repository can be installed as a GitHub action step:

[image]

However, I get an error when using this which indicates to me that it is not published as action:

[image]

And I can't see it on the marketplace:

[image]

Has it been unpublished since the documentation was written?

nerdyscout, 2021-05-02:
actually it was never published since implemented.
As this feature was never completely tested you might want to use instead [kicad-exports](https://github.com/nerdyscout/kicad-exports) which uses kibot v0.11.0

set-soft, 2021-05-04:
Hi @jameswilddev and @nerdyscout !

What @nerdyscout says is true, we never finished the integration. This is why it isn't published.

But the problem was different (missleading example). Now I added a useful tag name "v1". Updated the docs and also fixed a detail in the `Dockerfile` (a small detail about `curl` package, now available from backports).

A working example can be found here: https://github.com/INTI-CMNB/kicad-ci-test-spora/tree/test_gh_action

closed 2021-05-04 by set-soft