INTI-CMNB/KiBot #73: per-project preflight filter that applies too all runs?(bundles)

yanghu, 2021-05-06:
Is it possible to define a filter for preflight drc/erc errors for a project?

The use case would be: 
I have a bundle that I'd like to run on a project

However, I want to ignore some drc errors. 

Now, I need to modify every yaml file that bundle includes, and update the filters section in each yaml file. This means I need a new set of yaml files for this project only to get a new filtering behabior.

It would be nice if I could define a filter somewhere for the project, and each workflow in that bundle would load the filter and use it for preflight checks. Then I can reuse the same bundle and yaml files for my projects that has different filtering needs.

yanghu, 2021-05-06:
seems like bundle is the feature introduced in https://github.com/nerdyscout/kicad-exports on top of kibot. I'll file the request in that repo instead.

closed 2021-05-06 by yanghu

set-soft, 2021-05-06:
Hi @yanghu !

Also note that you can include outputs from other config files: https://github.com/INTI-CMNB/KiBot#importing-outputs-from-another-file

Here is an example of a configuration file that generates fabrication files for 5 known manufacturers: https://github.com/INTI-CMNB/kicad-ci-test-spora/blob/master/pcb_main/spora.kibot.yaml

I didn't apply it to preflights because I think the filters are very specific to a project. What do you want to exclude in various projects?