INTI-CMNB/KiBot #76: Wishlist: Automated annotation of power components

bene42, 2021-05-18:
Power components are annotated by kicad automagically.
This impede change tracking in a SCM dramatically. As a common workaround
of this unrelated changes in the schematic file filters can be used. In git-SCM
this filter looks like:

```
[filter "kicad_sch"]
        clean = "sed -E 's/#(PWR|FLG)[0-9]+/#\\1?/'"
        smudge = cat
```
Unfortionaly there is a side effect related to kibot:
Running kibot on a fresh checkout may end up in a failure due to missing annotations.

set-soft, 2021-05-19:
Hi @bene42 !

What exactly fails? 

I never had problems with PWR or FLG components. They get annotated with all the rest and remains annotated. Of course is quite annoying to discover that some of them aren't annotated, because you don't see the annotation.

jnavila, 2022-01-03:
I also use the described git filter, so that the schematic files are checked in version control with `PWR?` and `FLG?` annotations. This should not be an issue most of the time, but kibot does not like it.

commit 27c98721 (2022-02-26): New preflight to annotate power components

Closes #76
closed 2022-02-26 by set-soft