INTI-CMNB/KiBot #77: Failed to import pcbnew Python module without using virtual env

besi, 2021-05-24:
Hello, 

I want to use `KiBot` on my Mac. I am not using python virtual envs as described in #39. However, I'm running into the same error:

> ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to `PYTHONPATH`? (kibot - ?`__main__.py:222`)

Here is what I did in order to try to mitigate the problem:

    # locate the pcbnew.py file inside the KiCad.app
    ls /Applications/KiCad/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.py
    /Applications/KiCad/kicad.app/Contents/Frameworks/python/site-packages/pcbnew.py

    # add this path to the PATH as well as to the PYTHONPATH as mentioned in the error message
    export PYTHONPATH="$PYTHONPATH:/Applications/KiCad/kicad.app/Contents/Frameworks/python/site-packages"      
    export PATH="$PATH:/Applications/KiCad/kicad.app/Contents/Frameworks/python/site-packages"                

    # It's still not working
    kibot
    ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to PYTHONPATH? (kibot - __main__.py:222)

What is peculiar is the fact that there is a `pcbnew_do` binary:

    pcbnew_do --version
    pcbnew_do 1.5.8 - Copyright 2018-2021, INTI/Productize SPRL - License: Apache 2.0

Locate `python`:

    which python   
    /usr/local/bin/python
    python --version
    Python 2.7.17

Locate pythons symlink:

    ls -lia `which python` 
    31241519 lrwxr-xr-x  1 besi  admin  36 Nov 24  2019 /usr/local/bin/python -> ../Cellar/python@2/2.7.17/bin/python

set-soft, 2021-05-24:
Hi @besi !

Lamentably we currently don't support Python 2.7 (declared obsolete long time ago). Adding support is a really complex task that I won't do (unless I have a real incentive).

What's even more lamentable is that KiCad 5.x is still using Python 2.7 on Windows and MacOSX.

You most probably have Python 3.x on your system and you can run KiBot, but when KiBot tries to load `pcbnew` (basically the whole KiCad encapsulated in a 30 MiB lib, ugly!) fails. This is because you have a Python 2.7 `pcbnew` and not a Python 3.x one.

See more details on #20 

This problem should be fixed when KiCad 6 is finally ready and we migrate to it.

closed 2021-05-24 by set-soft

besi, 2021-05-25:
Ok thanks for the heads up. Just that I get you right. There is currently no way to run `KiBot` on my mac?

set-soft, 2021-05-25:
Well ... maybe running a VirtualBox with Linux inside.

set-soft, 2021-05-27:
Any doc enhancement is welcome. Take a look at what the docs says and add all you find useful.

besi, 2021-05-27:
Well on Github I did not get far. Using Python 2.7 I could not install the required dependencies:
https://github.com/besi/godfatherphone-hat/runs/2687164735?check_suite_focus=true

set-soft, 2021-05-27:
You don't need to do it just use a docker image.
Take a look at this project: https://github.com/INTI-CMNB/kicad-ci-test-spora
In this project you'll find an explanation of how to apply KiBot to a repo on GitHub.
You can also use GitLab, the docs about CI/CD are here: https://github.com/INTI-CMNB/KiBot#usage-for-cicd

Gasman2014, 2021-12-04:
KiCad6 on macOS now packages Python 3.8.2 but this issue still occurs. 

I have set my `$PYTHONPATH` to the 3.8.2 executable. `pcbnew` seems to be loaded as I get valid output to `'help("pcbnew")'` but get the same error. 

Shall I open this as a new issue?

```
echo $PYTHONPATH
/Applications/KiCad/KiCad.app/Contents/Frameworks/Python.framework/Versions/3.8/bin/python3.8

 %  $PYTHONPATH
Python 3.8.2 (v3.8.2:7b3ab5921f, Feb 24 2020, 17:52:18)
[Clang 6.0 (clang-600.0.57)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> help("pcbnew")

Help on module pcbnew:

NAME
    pcbnew

DESCRIPTION
    # This file was automatically generated by SWIG (http://www.swig.org).
    # Version 4.0.2
    #
    # Do not make changes to this file unless you know what you are doing--modify
    # the SWIG interface file instead.

CLASSES
    builtins.object
        BASE_SET
            LSET
        BOARD_DESIGN_SETTINGS
        BOARD_LISTENER
        CLIPPER_Z_VALUE
        CN_DISJOINT_NET_ENTRY
        CN_ZONE_ISOLATED_ISLAND_LIST
        COLOR4D
        CONNECTIVITY_DATA
        D356_RECORD
<snip>

 %  kibot
ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to PYTHONPATH? (kibot.kibot.kiplot - kiplot.py:43)
ERROR:Failed to import pcbnew Python module. Is KiCad installed? Do you need to add it to PYTHONPATH? (kibot.kibot.kiplot - kiplot.py:43)
```

set-soft, 2021-12-04:
Hi @Gasman2014 !
KiCad 6 isn't supported yet.
The PYTHONPATH environment variable should point to the place where modules are installed, not to the executable. Here the [documentation](https://docs.python.org/3/using/cmdline.html#envvar-PYTHONPATH)
You should try `import pcbnew` not just getting the help

Gasman2014, 2021-12-05:
Thanks - didn't appreciate that KiCad6 wasn't supported - yet ... ?
Thanks for the hint on the PYTHONPATH.
Yeah, I just showed the help to demonstrate that it was installed and seen but understand it needs to be imported.

set-soft, 2021-12-07:
> Thanks - didn't appreciate that KiCad6 wasn't supported - yet ... ?

KiCad 6 was developed with new features as a goal, not compatibility, and some key features in the v6 roadmap (that could impact a lot) were simply dropped. So v6 support isn't simple.

> Thanks for the hint on the PYTHONPATH.
> Yeah, I just showed the help to demonstrate that it was installed and seen but understand it needs to be imported.

Ok, but please try to import it.

You could try KiAuto first, is currently supporting v6.0.0 rc1 (no 3D view, because KiCad simply dies if no OpenGL is available, which is the case for a Virtual X server).  [KiAuto](https://github.com/INTI-CMNB/KiAuto) provides some of the key features of KiBot (ERC, DRC, print, etc.)