INTI-CMNB/KiBot #78: GS.sch is None when running with Kicad nightly (5.99)

justyn, 2021-06-03:
I'm trying to update my configuration to run with Kicad nightly (5.99).

I've set the environment variable `KIAUS_USE_NIGHTLY=5.99`, which allows the DRC to run.

```
kibot -c gen-outputs.yaml -d output/build -e testfile.kicad_sch -b testfile.kicad_pcb
```

Yet most of the other tools seem to error because `GS.sch` is `NoneType`.

```
- Running the DRC
- 'Generate Gerber X2.' (gerber) [gerber]
Traceback (most recent call last):
  File "tools/kicad-jb-make/tools/KiBot/src/kibot", line 16, in <module>
    main()
  File "/mnt/files/pro/mir/pcb/stem/std/tools/kicad-jb-make/tools/KiBot/kibot/__main__.py", line 358, in main
    generate_outputs(outputs, args.target, args.invert_sel, args.skip_pre)
  File "/mnt/files/pro/mir/pcb/stem/std/tools/kicad-jb-make/tools/KiBot/kibot/kiplot.py", line 364, in generate_outputs
    run_output(out)
  File "/mnt/files/pro/mir/pcb/stem/std/tools/kicad-jb-make/tools/KiBot/kibot/kiplot.py", line 340, in run_output
    out.run(get_output_dir(out_dir))
  File "/mnt/files/pro/mir/pcb/stem/std/tools/kicad-jb-make/tools/KiBot/kibot/out_any_layer.py", line 256, in run
    self.options.run(output_dir, self.layers)
  File "/mnt/files/pro/mir/pcb/stem/std/tools/kicad-jb-make/tools/KiBot/kibot/out_any_layer.py", line 125, in run
    super().run(output_dir)
  File "/mnt/files/pro/mir/pcb/stem/std/tools/kicad-jb-make/tools/KiBot/kibot/out_base.py", line 295, in run
    comps = GS.sch.get_components()
AttributeError: 'NoneType' object has no attribute 'get_components'
```

Any idea what could be causing GS.sch to turn up as `None`?

Thanks.

set-soft, 2021-06-03:
None of the outputs involving components is working with KiCad 5.99.

The format for the schematic changed completly (and using a rare format, the same used for the PCB, not a common format like XML or JSON/YAML).  The roadmap for KiCad 6 says that a Python API for EEschema will be added. So I didn't write a parser for the new format, waiting for the API to be available.

The API isn't currently available.

KiCad 6 will be quite incompatible with KiCad 5 in many aspects. And the developers doesn't want to provide an easy migration path for the plug-ins. So you'll have to be patient. IMHO this is a huge difference between the current state of most free software projects and proprietary tools, and this includes major components like the Linux Kernel or the GNOME environment. You don't have mature and stable APIs, they are poorly documented and changing all the time. The Python API for KiCad 6 will introduce a lot of incompatibilities. Some of them are already coded in KiBot, but not all. As the API changes all the time I'm not udating KiCad 5.99 support all the time. Otherwise I should stop adding new features.

Conclusion: Unless you want to help to support KiCad 5.99 I'll suggest to wait until KiCad 6 is out and we migrate to it.

closed 2021-06-03 by set-soft

justyn, 2021-06-03:
Hi @set-soft

I see, I had seen the references to Kicad 6 in the documentation and wrongly assumed it was supported already.

Certainly if a good Python API can provide a more standard way of scripting these tasks hopefully it will be better in the long run.

Thanks for your work on this, and the explanation.