INTI-CMNB/KiBot #79: Managing tolerances

thejpster, 2021-06-18:
What is the recommendation for specifying tolerances on KiCad parts - e.g. for capacitors or resistors?

Currently I have a "Tolerance" field on each resistor and capacitor, e.g. "10%" or "X5R". Capacitors also have a "Voltage" field, e.g. "16V".

I have set the voltage and tolerance for each part to be the minimum suitable for that design - although when selecting parts, "better" is generally fine.

When I build a BoM though I get warnings like the following:

```
WARNING:(W004) Field conflict: (C207 C210 C211 C212 C215 C302 C401 C402 C403 C404 C803 C806 C903 C1001 C1002 C1101 C1102) [voltage] : '50V' <- '16V' (in C401) (kibot.kibot.bom.bom - log.py:82)
WARNING:(W004) Field conflict: (C207 C210 C211 C212 C215 C302 C401 C402 C403 C404 C803 C806 C903 C1001 C1002 C1101 C1102) [voltage] : '50V 16V' <- '10V' (in C403) (kibot.kibot.bom.bom - log.py:82)
WARNING:(W004) Field conflict: (C224 C807 C1105) [voltage] : '25V' <- '10V' (in C807) (kibot.kibot.bom.bom - log.py:82)
WARNING:(W004) Field conflict: (C202 C203 C204 C206 C208 C209 C213 C214 C216 C217 C405 C804 C808) [voltage] : '16V' <- '10V' (in C405) (kibot.kibot.bom.bom - log.py:82)
```
In some cases the MPN for a component specifies a higher voltage rating because that is all that is available. Is KiBot somehow finding the voltage in the MPN and getting upset that it doesn't match the "Voltage" field? If so, how can I stop it from doing that? Or is this because I have multiple parts with the same "Value" and the same "Footprint", but with different values for "Voltage"?

thejpster, 2021-06-18:
No sooner as I posted this, I found the `group_fields` attribute, which tells the BOM tool which fields to distinguish components on.

Maybe 'Voltage', 'Tolerance' and 'Current' should be added to the default list?

set-soft, 2021-09-27:
Hi @thejpster !
Sorry for the huge delay. I wasn't sure when I read it and then I forgot it.
I makes sense to separate components with different tolerance, voltage, current and/or power, so I added these names to the list.
Thanks!

closed 2021-09-27 by set-soft