INTI-CMNB/KiBot #81: Unknown filter type: `rot_footprint`

aep, 2021-07-27:
when installing using pip

   https://github.com/INTI-CMNB/KiBot#installation-using-pip

rot_footprint doesn't appear to be available.

```
ERROR:Unknown filter type: `rot_footprint` (kibot.kibot.error - error.py:40)

```

```
$ kibot --version
KiBot 0.8.1 - Copyright 2018-2020, Salvador E. Tropea/INTI/John Beard - License: GPL v3+
```

aep, 2021-07-27:
apparantly you can just use src/kibot from the git clone.

however, that one doesnt have  variant: rotated

```
ERROR:In section 'position' (position): Unknown variant name `rotated` (kibot.error - error.py:40)
```

closed 2021-07-27 by aep

set-soft, 2021-07-30:
The last version on PyPi is 0.11.0, not 0.8.1.
You can also "pip install" from a git repo. Is something like "git+REPO".
And yes, you can run KiBot directly from the repo, no need to install. Dependencies must be met, of course.