INTI-CMNB/KiBot #83: Missing components of a custom library in schematic PDF output when using simple variant

myorbit, 2021-08-13:
This could be similar to issue #82.

In our project we use custom libraries with a global library scope.

The default schematic PDF output is OK and all components are shown.

```yaml
outputs:
  - name: 'print_sch'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: .
```

However, if I add a 'simple' variant by using a filter to exclude some components, so they can be marked with a red cross in the schematic print, all the components from our custom library are missing in the the schematic PDF output.

```yaml
filters:
 - name: 'by prefix'
   type: 'generic'
   comment: 'Remove components by prefix'
   exclude_refs:
     - FL1
     - U23
     - C75
     - C76
 
outputs:
  - name: 'print_sch_variant'
    comment: "Print schematic (PDF)"
    type: pdf_sch_print
    dir: .
    options:
      dnf_filter: 'by prefix'
```

To summarize when using this 'simple' variant style:
* not excluded component from KiCad library: visible, no red cross
* not excluded component from custom library: missing, no red cross
* excluded components from KiCad library: visible, red cross
* excluded components from custom libray: missing

When I check the relevant output of the GitLab pipeline this is what happens...

... for 'print_sch':

```
$ kibot -d Fabrication -s run_drc,run_erc print_sch
Using config file: kicad_ci_test.kibot.yaml
WARNING:(W044) More than one SCH file found in current directory.
  Using prototype.sch if you want to use another use -e option. (kibot - log.py:82)
- Updating BoM in XML format
WARNING:(W058) Missing KiCad main config file /root/.config/kicad/kicad_common (eeschema_do - misc.py:161) (kibot.kibot.kiplot - log.py:82)
WARNING:(W041) Missing library `dvtk_Device` (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W041) Missing library `dvtk_Linear-Amplifiers-Instrumentation-OP-Amps-Buffer-Amps` (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W041) Missing library `dvtk_Power_Management` (kibot.kibot.kicad.v5_sch - log.py:82)
- 'Print schematic (PDF)' (print_sch) [pdf_sch_print]
Found 5 unique warning/s (5 total)
Uploading artifacts for successful job 00:03
Uploading artifacts...
```

... for 'print_sch_variant':

```
$ kibot -d Fabrication -s run_drc,run_erc print_sch_variant
Using config file: kicad_ci_test.kibot.yaml
WARNING:(W044) More than one SCH file found in current directory.
  Using prototype.sch if you want to use another use -e option. (kibot - log.py:82)
- Updating BoM in XML format
WARNING:(W058) Missing KiCad main config file /root/.config/kicad/kicad_common (eeschema_do - misc.py:161) (kibot.kibot.kiplot - log.py:82)
WARNING:(W041) Missing library `dvtk_Device` (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W041) Missing library `dvtk_Linear-Amplifiers-Instrumentation-OP-Amps-Buffer-Amps` (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W041) Missing library `dvtk_Power_Management` (kibot.kibot.kicad.v5_sch - log.py:82)
- 'Print schematic (PDF)' (print_sch_variant) [pdf_sch_print]
WARNING:(W043) Missing component `dvtk_Device:ESDA7P60-1U1M` (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W043) Missing component `dvtk_Device:744232090` (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W043) Missing component `dvtk_Linear-Amplifiers-Instrumentation-OP-Amps-Buffer-Amps:TLV2333IDGKx` (kibot.kibot.kicad.v5_sch - log.py:82)
WARNING:(W043) Missing component `dvtk_Power_Management:TPS2112A` (kibot.kibot.kicad.v5_sch - log.py:82)
Found 9 unique warning/s (18 total)
Uploading artifacts for successful job 00:03
Uploading artifacts...
```

Should I add the components as a Git submodule? Which scope should I use for the libraries then in the KiCad project? Does this happen when using a 'normal' variant?

set-soft, 2021-08-18:
Hi @myorbit !
Your problem seems to be realted to the lack of the libraries in the repo. As you suggest they could be a git submodule.

The libraries must be accesible, I think you don't have problems with `print_sch` because you have a cache lib (`*-cache.lib`).

I'm not sure what do you mean with *global library scope*, do you have the libaries in the same directory where system libs are installed?

Do you have a `sym-lib-table` file? What does it contain? Are the paths declared there available in the docker image?

I always use relative paths and include the libraries in the repo. Like this:

```
(sym_lib_table
  (lib (name tedy_library)(type Legacy)(uri ${KIPRJMOD}/../bibliotecas/tedy_library.lib)(options "")(descr ""))
)
```

The `KIPRJMOD` variable expands to the place where the project is located. So here I say the lib is in `../bibliotecas` and is named `tedy_library.lib`. In the schematic this lib is referenced as `tedy_library`.

If you use a git submodule be sure the CI/CD does a checkout of your project **and** the submodules.

Let me know if you fix it this way.

set-soft, 2021-08-24:
Hi @myorbit !

The last patch should add support for caches, so you don't need to include the libs.
Please let me know if it solves your problem

closed 2021-08-24 by set-soft

set-soft, 2021-10-07:
Hi @myorbit !
I recently created a *development* docker image, the tag is `dev`, pulling `setsoft/kicad_auto:dev` you get the last code that passed the regression tests. 
BTW: It could help if you mention what problems do you have using Arch Linux. I only test the code on Debian, and it has some particular details, isn't a pure Python environment (which is good because Python core developers are really alien to UNIX systems).

commit 40a42513 (2021-08-24): Added suport for schematic cache files when components are missing
Fixes #83