INTI-CMNB/KiBot #85: Add support for saving updated PCB after drc check

danecreekphotography, 2021-09-19:
I'm working on integrating KiBot into a full CI/CD release flow using GitHub automation, including setting the version of the schematic and PCB to the GitHub release version.

I have things set up to change the version on the schematic before the PDF gets generated, and I can set the version in the .pcb file as well, however this will only have an effect on versions included in silkscreen layers. It won't affect version text that might be on the copper layer.

According to the KiAuto documentation:

> Refilling copper zones

> When you run the DRC KiCad will refill all zones. If you didn't do it before saving it could lead to a situation where the PCB that passes DRC isn't the one saved to disk. To solve you can use -s option to save the PCB after DRC:

> pcbnew_do run_drc -s YOUR_PCB.kicad_pcb DESTINATION/

It doesn't look like this option is exposed by KiBot however. It'd be great if there were a flag to pass to KiBot to say "save the updated PCB file from DRC". This would enable setting the version in the `.kicad_pcb` file before doing DRC, having the zones refill, then I could overwrite the original `.kicad_pcb` and KiBot could continue generating outputs using the updated PCB.

danecreekphotography, 2021-09-19:
https://github.com/INTI-CMNB/KiBot/blob/0fc1c11c50280c4f5f32a21596332ceb0a187698/kibot/pre_drc.py#L46

Looks like the following changes would be required:

1. Test against a new option (`persist_fills`?)
2. If present append the `-s` option

danecreekphotography, 2021-09-19:
Never mind. I should have checked before I opened this, the copper fill around the text area is plenty big enough to not need to be re-generated after changing the version number.

closed 2021-09-19 by danecreekphotography

set-soft, 2021-09-20:
Hi @danecreekphotography, if you need it in the future take a look at the `check_zone_fills` preflight option. This will do a zone fill before any action that needs it.

danecreekphotography, 2021-09-20:
@set-soft Thanks! Will that persist the updated PCB with the newly filled zones?

set-soft, 2021-09-20:
@danecreekphotography this option will ensure the areas are refilled. The PCB isn't saved to disk, instead KiCad is asked to refill the zones before every operation that could be affected. Currently this is:
1) Before "printing"
2) Before "plotting" (i.e. generate the gerbers, export the DXF, etc.)

danecreekphotography, 2021-09-20:
Excellent, thank you. That's exactly what I need!