INTI-CMNB/KiBot #86: ERC check times out on schematics with hierarchical sheets

danecreekphotography, 2021-09-20:
I've successfully got KiBot ERC checks running with a schematic that consists of only a single sheet, but it's failing every time I try to run it with a schematic that has hierarchical sheets. The error is the same every time:

```
WARNING:(W054) Time out detected, on slow machines or complex projects try: (kibot.kibot.kiplot - log.py:82)
WARNING:(W054) `kiauto_time_out_scale` and/or `kiauto_wait_start` global options (kibot.kibot.kiplot - log.py:82)
```

I know it's running ERC on the sub sheet because I have an ERC warning getting returned from it:

```
WARNING:(W040) Component H? (MountingHole) is not annotated (kibot.kibot.kicad.v5_sch - log.py:82)
```

I've tried adding the following to my .kibot.yml file but it didn't make a difference:

```yaml
global:
  kiauto_time_out_scale: 2.0
```

The command line I'm running:

```bash
 kibot -c .kibot/release.kibot.yaml -b TBM930_De-ice_panel.kicad_pcb --out-dir KiBotOutput
```

The KiBot file is available [in my GitHub repo](https://github.dev/danecreekphotography/TBM930_De-ice_panel/blob/1824a13ce2e98e09094641cea6e2a1c3cf5988f0/.kibot/release.kibot.yaml#L1-L1).

set-soft, 2021-09-21:
Try running it with full debug information: `-vvv`
It will print a lot of information and also create `run_erc_eeschema_screencast.ogv` in the output dir.

It runs on my system:

```
salvador@kotori:~/.../0Test/TBM930_De-ice_panel$ kibot  -c .kibot/release.kibot.yaml -b TBM930_De-ice_panel.kicad_pcb --out-dir KiBotOutput
- Running the ERC
- Running the DRC
- 'PDF - print schematic' (print_sch) [pdf_sch_print]
- 'PDF - front copper and user drawings' (print_front) [pdf_pcb_print]
- 'PDF - back copper and user drawings' (print_bottom) [pdf_pcb_print]
- 'Interactive Bill of Materials' (interactive_bom) [ibom]
- 'Renders - top (white)' (pcb_top_w) [pcbdraw]
- 'Renders - bottom (white)' (pcb_bot_w) [pcbdraw]
- 'Renders - 3D model (STEP)' (step) [step]
- 'Gerbers - JLCPCB' (JLCPCB_gerbers) [gerber]
- 'Drill - JLCPCB' (JLCPCB_drill) [excellon]
- 'Pick and place - JLCPCB' (JLCPCB_position) [position]
- 'BOM - JLCPCB' (JLCPCB_bom) [bom]
- 'Gerber and drill zip - JLCPCB' (JLCPCB) [compress]
- 'Gerbers - PCBWay' (PCBWay_gerbers) [gerber]
- 'Drill - PCBWay' (PCBWay_drill) [excellon]
- 'Gerber and drill zip - PCBWay' (PCBWay) [compress]
```

danecreekphotography, 2021-09-21:
I reset the repo last night to get rid of the sub sheets since I needed it in working order. I'll try again with -vvv this morning and a sub sheet and let you know what it says.

danecreekphotography, 2021-09-21:
Here's a branch with a subsheet: https://github.com/danecreekphotography/TBM930_De-ice_panel/tree/subsheet_test. I thought it might be an issue with spaces in the subsheet filename but it doesn't seem to make a difference.

Here's the command that runs and times out:

```
- Running the ERC
DEBUG:Executing: ['eeschema_do', '-r', '-vvv', 'run_erc', '-o', '/workspaces/TBM930_De-ice_panel/KiBotOutput/TBM930_De-ice_panel-erc.txt', '/workspaces/TBM930_De-ice_panel/TBM930_De-ice_panel.sch', '/workspaces/TBM930_De-ice_panel/KiBotOutput'] (kibot.kibot.kiplot - kiplot.py:164)
```

It looks like it times out waiting for the save dialog:

```
> INFO:Waiting for "ERC File save dialog" ...
> DEBUG:Closing pipe with 6685 (eeschema_do.kiauto.ui_automation - ui_automation.py:41)
> DEBUG:Terminating 6685 (eeschema_do.kiauto.ui_automation - ui_automation.py:53)
> DEBUG:Terminating the session recorder (eeschema_do.kiauto.ui_automation - ui_automation.py:161)
> DEBUG:Closing pipe with 6684 (eeschema_do.kiauto.ui_automation - ui_automation.py:41)
> DEBUG:Terminating 6684 (eeschema_do.kiauto.ui_automation - ui_automation.py:53)
> Traceback (most recent call last):
>   File "/usr/bin/eeschema_do", line 620, in <module>
>     eeschema_run_erc_schematic(cfg)
>   File "/usr/bin/eeschema_do", line 352, in eeschema_run_erc_schematic
>     eeschema_run_erc_schematic_5_1(cfg)
>   File "/usr/bin/eeschema_do", line 282, in eeschema_run_erc_schematic_5_1
>     wait_for_window('ERC File save dialog', 'ERC File')
>   File "/usr/lib/python3/dist-packages/kiauto/ui_automation.py", line 344, in wait_for_window
>     raise RuntimeError('Timed out waiting for %s window' % name)
> RuntimeError: Timed out waiting for ERC File save dialog window
> DEBUG:Restoring old eeschema config (eeschema_do.kiauto.file_util - file_util.py:152)
> DEBUG:Restoring old KiCad common config (eeschema_do.kiauto.file_util - file_util.py:152)
> DEBUG:Checking if /workspaces/TBM930_De-ice_panel/TBM930_De-ice_panel.pro was modified (eeschema_do.kiauto.file_util - file_util.py:242)
>  (kibot.kibot.kiplot - kiplot.py:175)
WARNING:(W054) Time out detected, on slow machines or complex projects try: (kibot.kibot.kiplot - log.py:82)
WARNING:(W054) `kiauto_time_out_scale` and/or `kiauto_wait_start` global options (kibot.kibot.kiplot - log.py:82)
ERROR:ERC returned 1 (kibot.kibot.pre_erc - pre_erc.py:65)
The terminal process "bash '-c', 'kibot -vvv -c .kibot/release.kibot.yaml -b TBM930_De-ice_panel.kicad_pcb --out-dir KiBotOutput'" terminated with exit code: 10.
```

danecreekphotography, 2021-09-21:
Here you go!
[logs.zip](https://github.com/INTI-CMNB/KiBot/files/7203214/logs.zip)

set-soft, 2021-09-21:
The problem is quite simple and related to:
```
WARNING:(W040) Component H? (MountingHole) is not annotated (kibot.kibot.kicad.v5_sch - log.py:82)
```
KiCad won't run an ERC on a project that isn't fully annotated.

closed 2021-09-21 by set-soft

danecreekphotography, 2021-09-21:
@set-soft Is it possible to get KiBot to print out a better error message for this situation than a timeout error?

The reason I'm running ERC/DRC during pull requests is to prevent exactly this kind of error from getting checked in. It'd be clearer in the build system if the failure was "ERROR: Components must be fully annotated" instead of "WARNING:(W054) Time out detected, on slow machines or complex projects try: (kibot.kibot.kiplot - log.py:82)"

set-soft, 2021-09-21:
I can make the warning more descriptive, but I can't make it an error. 
The schematic is loaded using my own implementation of the SCH format. When the code detects a component isn't annotated I think KiCad will fail, but I can't be sure. This is why I generate a warning. 
I'll try to catch the following situatiion:

1) We detected an annotation error
2) The ERC failed with time-out

In this case I'll add a message explaining the most probable reason.

set-soft, 2021-09-21:
Now the output is:
```
WARNING:(W040) Component H? (MountingHole) is not annotated (kibot.kicad.v5_sch - v5_sch.py:1063)
- Running the ERC
WARNING:(W054) Time out detected, on slow machines or complex projects try: (kibot.kiplot - kiplot.py:179)
WARNING:(W054) `kiauto_time_out_scale` and/or `kiauto_wait_start` global options (kibot.kiplot - kiplot.py:180)
ERROR:ERC returned 1 (kibot.pre_erc - pre_erc.py:65)
ERROR:Make sure your schematic is fully annotated (kibot.pre_erc - pre_erc.py:67)
```

danecreekphotography, 2021-09-21:
Nice! Thank you!