INTI-CMNB/KiBot #87: Wrong origin for Pick & Place file

fleborne, 2021-09-24:
Hello,

I have a problem with the Pick & Place file: the positions are given relatively to the global origin, and I don't find how to use the auxiliary axis origin.

In the documentation of KiBot, it says that the expected result is the same as for running File > Fabrication Outputs > Footprint Position (.pos) File. This is not what I observe, as using KiCad's built-in script yields the correct result (components placed relative to the auxiliary axis origin) while using KiBot's script results in components being placed relative to the global origin.

set-soft, 2021-09-27:
Hi @fleborne !
You are right, KiCad always uses the auxiliar origin.
Now we default to it, but you can choose to use absolute coordinates.
Please try the current git code, I just tested a simple case and verified a couple of coordinates.
Thanks!

closed 2021-09-27 by set-soft

fleborne, 2021-09-27:
Hi @set-soft,
Version 0.11.0 works for the board I'm working on :+1: 
Thanks a lot!

commit a31002ed (2021-09-27): Fixes position files, now defaults to use the auxiliar origin as KiCad.

Can be disabled to use absolute coordinates.
Fixes #87