INTI-CMNB/KiBot #91: kicad2step can't find 3D models if KISYS3DMOD variable is not set

kosma, 2021-10-05:
When generating STEP outputs, the KISYS3DMOD variable from KiCad config fileis not passed to `kicad2step`, resulting in all models from the system library missing in the generated output. I've verified the issue by calling `kicad2step` manually; it can't find the system 3D models unless it has the env variable set.

Here's a quick and un-elegant hack that fixed the issue for me (`out_step.py`):

```diff
     def download_models(self):
         """ Check we have the 3D models.
             Inform missing models.
             Try to download the missing models """
         models_replaced = False
         # Load KiCad configuration so we can expand the 3D models path
         KiConf.init(GS.pcb_file)
+        os.environ['KISYS3DMOD'] = KiConf.kicad_env['KISYS3DMOD']
         # List of models we already downloaded
         downloaded = set()
         self.undo_3d_models = {}
```

kosma, 2021-10-12:
A more specific test:

I've ran strace on the thing: `strace -s64 -fv -otrace kicad2step -f -o test.step *.kicad_pcb`. There's no mention of `KISYS3DMOD`, `packages3d` or `kicad_common` in the trace. Resulting file is missing the system models. The only interesting mention if the word `kicad` is this:
```
stat("/home/kosma/.config/kicad", ...) = 0
stat("/home/kosma/.config/kicad/3d", 0x7ffcdca28520) = -1 ENOENT (No such file or directory)
```

set-soft, 2021-10-14:
Hi @kosma !
I added some code to catch:

1) Missing `KISYS3DMOD` definition 
2) Missing `3d` dir

In both cases the code tries to figure out the correct value for `KISYS3DMOD` and exports it in the environment

This should solve your problem, let me know if the code needs adjustements.

BTW: I never test KiCad 3D models installed in the CI/CD environment because:

1) They are huge (about 5 GiB)
2) My experience is that any project intended to last some years needs a copy of the symbols, modules and 3D models that is using. Otherwise you risk to experiment problems because KiCad changes them. 

If you take a look at the test cases they just download the needed models from KiCad's repo, this is what KiBot does if the models aren't installed.