INTI-CMNB/KiBot #100: Do not consider differences in the schematic's variant fields as a warning

mdeweerd, 2021-11-12:
Variants can be different for components of the same type.
(comments too in practice).

This warning is unexpected:
```
WARNING:(W004) Field conflict: (J1,J2,J3) [variant] : 'conn2' <- 'conn3' (in J3) (kibot.kibot.bom.bom - log.py:82)
```

set-soft, 2021-11-15:
Hi @mdeweerd !
You are right, but this is a little bit tricky because we support various `variants` mechanisms.
This patch should fix it, but I don't have a test case at hand. Please consider providing simple test cases for your reports.

closed 2021-11-15 by set-soft

set-soft, 2021-11-15:
> I'll use the demo project for test cases when needed - ok ?

This is ok if this is simpler for you.

set-soft, 2021-11-15:
The test are run at GitHub after each push.

You can see how the GitHub servers runs it here: https://github.com/INTI-CMNB/KiBot/blob/master/.github/workflows/pythonapp.yml

It uses a special docker image: `setsoft/kicad_auto_test:latest` this image has the tools needed to run the tests.

The tests are run with `pytest-3 --test_dir output`

mdeweerd, 2021-11-15:
Thanks, I managed to set it up.  I had to map the 'kibot' dir to ${HOME}/kibot (not ${HOME}/workdir ).

This runs `test_misc` on Win10 with Cygwin:

```bash
#!/bin/bash
# Runs on Windows/Cygwin
IMAGE=setsoft/kicad_auto_test:latest
export USER_ID=$(id -u)
export GROUP_ID=$(id -g)
# Using root on windows
USER_ID=0
group_ID=0
export WORKDIR=$(realpath "$(dirname $0)/..")
export SUBDIR=$(basename $0)
export WORKDIR=$(realpath "$(dirname $0)")
export SUBDIR=.
export DISPLAY=${DISPLAY:=0}
export HOMEDIR=$(realpath "$(dirname $0)/..")/kibot
if [ ${USER_ID} != 0 ] ; then
  export TGTHOMEDIR="/home/${USER}"
else
  export TGTHOMEDIR="/root"
fi

# Quote arguments
ARGS=""
for i in "$@" ; do
    ARGS=$ARGS" '"${i/\'/\\\'}"'"
done

DISPLAY=host.docker.internal:0.0

mkdir -p $HOMEDIR
WORKDIR=$(cygpath -w "$WORKDIR")
HOMEDIR=$(cygpath -w "$HOMEDIR")

LOCAL_KICAD_DIR=$(cygpath 'C:\Program Files\KiCad\share\kicad')
LOCAL_KICAD_LIBDIR=$(cygpath -d -w "$LOCAL_KICAD_DIR/library")
LOCAL_KICAD_MODULESDIR=$(cygpath -d -w "$LOCAL_KICAD_DIR/modules")
LOCAL_KICAD_LIB_SETTING="--volume=${LOCAL_KICAD_LIBDIR}:/usr/share/kicad/library:ro --volume=${LOCAL_KICAD_MODULESDIR}:/usr/share/kicad/modules:ro"

if [ "$ARGS" == "" ] ; then
  ARGS=bash
fi
ARGS="python3-coverage erase ; python3-coverage run -a src/kibot ---help-outputs > /dev/null ; pytest --log-cli-level debug -k 'test_misc' --test_dir output ; python3-coverage report python3-coverage html -d output/htmlcov ; coveralls"
ARGS="pytest-3 --log-cli-level debug -k 'test_misc' --test_dir output"
docker run --rm -it -v /tmp/.X11-unix:/tmp/.X11-unix -e DISPLAY=$DISPLAY \
    -v $WORKDIR:${TGTHOMEDIR}/kibot \
    --env NO_AT_BRIDGE=1 \
    --workdir="${TGTHOMEDIR}" \
    $IMAGE /bin/bash -c "cd kibot/$SUBDIR ; $ARGS" \
| tee $1.log
```

commit ca5dfb42 (2021-11-15): Internal BoM: the field used for variants doesn't produce conflicts.

- Fixes: #100