INTI-CMNB/KiBot #99: 2D Image Generation missing available components

mdeweerd, 2021-11-12:
Using the `kibot_variants_arduprog` demo project, I added 2D renderings with components to the [yaml file (link to my fork)](https://github.com/mdeweerd/kibot_variants_arduprog/blob/d134f1a73a3bf50d23c415ffd957d212e34f8d4f/ardu_prog.kibot.yaml).  
I also mapped my local kicad library and modules to the docker image.

I then run `kicad  t1.pro` within the docker file with the host providing the XServer, and I get the following 3D View:
[image]

So all components are there, but when generating the 2D jpg using `kibot`, I only get the following image - its always the same regardless of whether I map my local KiCAD libraries or not:
[image]. 

I am not sure what the reason is - the demo repository has all 3D models locally, so there should not be a good reason for the missing components.

In my own project, I get 0 components and I do have several models in the project directory, and some models in the KiCAD library.  In the 3D View, I get the 3D models for the components in the project directory, not in KiCAD's library (but that's another issue that I can probably fix through a path configuration).

mdeweerd, 2021-11-12:
My 3D view was incomplete because I did not properly update my project's kibot wrapper.  It's ok now with this:
```
LOCAL_KICAD_DIR=$(cygpath 'C:\Program Files\KiCad\share\kicad')
LOCAL_KICAD_LIBDIR=$(cygpath -d -w "$LOCAL_KICAD_DIR/library")
LOCAL_KICAD_MODULESDIR=$(cygpath -d -w "$LOCAL_KICAD_DIR/modules")
LOCAL_KICAD_LIB_SETTING="--volume=${LOCAL_KICAD_LIBDIR}:/usr/share/kicad/library:ro --volume=${LOCAL_KICAD_MODULESDIR}:/usr/share/kicad/modules:ro"
```

The above maps the KiCad libraries on my host to the VM.  The 3D models are rendered as expected in the 3D view of the board using KiCad/PCBNEW in the container.

However, the 2D view is still showing an empty board.

kibot (out_pcbdraw) complains with quite a few message like this, and does so both for components inside the project (available in the container) and in the KiCAD libraries:

> Warning: component 'D_SOD-323' for footprint 'D2' from library 'Diode_SMD' was not found

I'll try to run pcbdraw interactively .

mdeweerd, 2021-11-12:
Lauching pcbdraw interactively.

1. Create styles `tmp1.json`:

 ```json
```

2. Run in the VM:

 I added strace to know where the tool is looking for files ;-).

```bash
./runInKiAuto bash -c 'apt update ; apt install strace ; strace -e file -ff pcbdraw -s /root/workdir/tmp1 --dpi 300 /root/workdir/STAR-LED-CC-DUO.kicad_pcb  /root/workdir/tmp2.svg' | tee pcbdraw.log
```

And the log reveals:
```raw
stat("/usr/lib/python3/dist-packages/pcbdraw/footprints/KiCAD-base/Diode_SMD/D_SOD-323.svg", 0x7ffcdf742260) = -1 ENOENT (No such file or directory)
stat("/root/.local/share/pcbdraw/footprints/KiCAD-base/Diode_SMD/D_SOD-323.svg", 0x7ffcdf742260) = -1 ENOENT (No such file or directory)
stat("/Diode_SMD/D_SOD-323.svg", 0x7ffcdf742260) = -1 ENOENT (No such file or directory)
Warning: component 'D_SOD-323' for footprint 'D2' from library 'Diode_SMD' was not found
```

I could find 'footprints' in /usr/share/pcbdraw/footprints.  There's not too many things there.

The 3D models are not used.

So the alternative is to export the 3D image from the 3D view.  What script can serve as an example for simulating an interactive procedure / how to capture the UI operations from an interactive session ?

```
```

mdeweerd, 2021-11-13:
I made a sample script to generate captures from the 3D Viewer:

[image]
[image]

Script: 
[gen3DCaptures.zip](https://github.com/INTI-CMNB/KiBot/files/7532676/gen3DCaptures.zip)

It's not perfect, but it provides a flow that would have to be integrated in KiAuto/src/pcbnew_do .

I'll open an issue on KiAuto for that.

nerdyscout, 2021-11-14:
afaik 2D renders are generated by using [pcbdraw](https://github.com/yaqwsx/PcbDraw/tree/master/pcbdraw) which uses its own [components library](https://github.com/yaqwsx/PcbDraw-Lib/tree/06b8c2cb8f62beffdd93d6f78c3223cd94d1b860/KiCAD-base) based on svg images. so the official 3d kicad library is not relevant here.

mdeweerd, 2021-11-14:
@nerdyscout I confirm that it's using SVG images, and they do not cover the entire library and not the new parts either - I verified this in https://github.com/INTI-CMNB/KiBot/issues/99#issuecomment-967217066 .

So using the 3D viewer is going to render more components.  It's not an orthogonal view, but close enough.

I didn't really find a tool that converts STEPs to SVGs (from a viewport) - maybe freecad could do it.  It might be ok for a top view, for a bottom view extra tricks are needed to hide what is normally hidden by the board.

The end result would not be close to this either (the BNC connector on the left is les hidden by the board than the right one.:
[image]

set-soft, 2021-11-15:
Hi @mdeweerd, as @nerdyscout correctly points out this is a PcbDraw weakness.

You can find some discussions (not a definitive solution) in the PcbDraw issues. 

Using FreeCAD you can generate SVG files for the top-view of STEP models, but they are big files.

I agree that a capture of the 3D viewer is an interesting solution. 

I'm closing this, but we can open another issue to discuss further details.

closed 2021-11-15 by set-soft

mdeweerd, 2021-11-15:
Yep, demo propose here: https://github.com/INTI-CMNB/KiAuto/issues/10 ;-).