INTI-CMNB/KiBot #101: Do not consider an empty DNP/dnp field as do not place - require '1'

mdeweerd, 2021-11-13:
I had a few components that were missing from the BOM generated under KiBot.
It turned out that there was an empty dnp field:
[image]

An empty dnp field should not be counted as a do not place value - it's only a dnp if the value is set to 1 (or not 0).

set-soft, 2021-11-15:
Thanks @mdeweerd !

It was implemented before we added this clarification to the docs.

I don't have a test case at hand, I just changed the regular expression to avoid filtering empty fields. Let me know if it works for you.

closed 2021-11-15 by set-soft

mdeweerd, 2021-11-15:
I'ld have to create a test case too, I adjusted the schematic files.  I'll probably create a few test cases ;-).

commit 087e36b2 (2021-11-15): KiCost variants: empty DNF fields shouldn't be excluded.

Fixes #101